homieiot/homie-esp8266 #98: make config.json settings readable

jasonkoon, 2016-06-04:
It would be nice to be able to use some/all configuration options during homie loops for example; Homie._interface.config->get().deviceId

This obviously doesn't work because _interface is private, but it would be nice to get at some of those options.

marvinroger, 2016-06-06:
It will land in 2.0.0, but I'm a bit curious, what would be your use case?

jasonkoon, 2016-06-06:
I just noticed unaiuars commit from 4 days ago to expose the id, so that fulfills my need.  I agree with others that it would be nice to be able to read/update those settings from the sketch level in response to certain MQTT messages.

My particular use case was to include the device ID in the MQTT payload.  The MQTT consumer wasn't set up to use a portion of the topic like I wanted, so I needed it as an entry in some JSON in the payload.

I can see wanting to change a wifi password and wanting to update all my Homie's to the new one OTA before making the change.

marvinroger, 2016-08-16:
Added in 9ae90b47a5ee6521b709da1b541ff14465fbb1f8

closed 2016-08-16 by marvinroger