homieiot/homie-esp8266 #101: LWT Topic is wrong (off by one?)

toke, 2016-06-15:
Affected Version v1.5.0 

## Problem

The last will topic is missing the last character `…/$online` -> `$onlin`

```
homie/devices/bb1/$online true
homie/devices/bb1/$nodes temperature:temperature,heatindex:heatindex,humidity:humidity
…
homie/devices/bb1/heatindex/degrees 27.38
homie/devices/bb1/$onlin false
```

## Expected result

The LWT should be complete `…/$online`

## How to reproduce

Using a arbitary sketch and compare the the LWT on boot and on connection loss.

Library version PubSubClient v2.6

## Notes

I've looked some minutes into it and tried to add a character on src/Homie/Boot/BootNormal.cpp:131 
`this->_fillMqttTopic(PSTR("/$online"));` like `  this->_fillMqttTopic(PSTR("/$onlineX"));` and it "solved" that issue in a way.

toke, 2016-06-15:
The problem seems to be related to something with platformio.
If I'm not too tired it works with the Arduino IDE.

marvinroger, 2016-06-18:
This is a weird one... To be honest, I am not using pio, but I don't see how this can happen.

marvinroger, 2016-06-19:
Anyway, v2.0.0 does not use the same MQTT implementation, and I just tested and it works perfectly. We'll then see if this happen again.

marvinroger, 2016-08-16:
I can't see this behavior, please reopen if you have an issue with the upcoming release.

closed 2016-08-16 by marvinroger