homieiot/homie-esp8266 #103: Compiling error - "does not name a type"

atschuni, 2016-06-22:
Hey there,

trying to use the examples provided by the homie-esp8266 lib. This is what i get when im trying to upload the sketch. What am I doing wrong?

```
Arduino: 1.6.9 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from /Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.hpp:11:0,
                 from /Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.h:4,
                 from /Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/examples/LightOnOff/LightOnOff.ino:1:
/Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootNormal.hpp:33:7: error: 'WiFiEventHandler' does not name a type
       WiFiEventHandler _wifiGotIpHandler;
       ^
/Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootNormal.hpp:34:7: error: 'WiFiEventHandler' does not name a type
       WiFiEventHandler _wifiDisconnectedHandler;
       ^
/Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootNormal.hpp:49:31: error: 'WiFiEventStationModeGotIP' does not name a type
       void _onWifiGotIp(const WiFiEventStationModeGotIP& event);
                               ^
/Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootNormal.hpp:49:58: error: ISO C++ forbids declaration of 'event' with no type [-fpermissive]
       void _onWifiGotIp(const WiFiEventStationModeGotIP& event);
                                                          ^
/Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootNormal.hpp:50:38: error: 'WiFiEventStationModeDisconnected' does not name a type
       void _onWifiDisconnected(const WiFiEventStationModeDisconnected& event);
                                      ^
/Users/Tom/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootNormal.hpp:50:72: error: ISO C++ forbids declaration of 'event' with no type [-fpermissive]
       void _onWifiDisconnected(const WiFiEventStationModeDisconnected& event);
                                                                        ^
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

```

marvinroger, 2016-06-22:
Download the [releases](https://github.com/marvinroger/homie-esp8266/releases), not the git development version. Also, use the documentation contained inside the `docs/` folder of the release you downloaded.

MaxChinni, 2016-06-28:
Just in case someone is interested in hacking on the latest git version and faces similar problems, here's what I did with my platformio project:

1. update platformio

        sudo pip install -U platformio
        platformio update
2. install the dependencies

        platformio lib install 64    # Json
        platformio lib install 1106  # Bounce2
        platformio lib install 346   # AsyncMqttClient
3. get the latest Homie git version

        git clone https://github.com/marvinroger/homie-esp8266.git lib/homie-esp8266

4. clean up the project and compile it

        platformio run -t clean
        platformio run

marvinroger, 2016-07-19:
Thanks @MaxChinni :)

closed 2016-07-19 by marvinroger