homieiot/homie-esp8266 #104: Error using subscribeToAll

mikecay, 2016-06-24:
Hi,

When I try to use the subscribeToAll function I get the following error.  

> class HomieNode' has no member named 'subscribeToAll'

I also tried the led strip example and I get the same error.  I noticed in the led example that the HomieNode construct did not have true as the forth parameter but, even after putting that in I still get the error.  

I am using Arduino IDE 1.6.8 and the forward declaration is in the code.  Does anyone have any idea?

Thank you,

Mike

marvinroger, 2016-06-24:
You're using [one of the releases](https://github.com/marvinroger/homie-esp8266/releases), right? Then you shoud use the docs contained inside the archive of your release. The documentation in the repo matches the git development version of Homie for ESP8266.

mikecay, 2016-06-24:
Thank you.  I was in fact using the examples on git rather that my release.  Got it now

Mike

marvinroger, 2016-06-24:
You're welcome!

closed 2016-06-24 by marvinroger