homieiot/homie-esp8266 #105: ESP-01 and GPIO0 not working

mikecay, 2016-06-25:
Hi

I have a sketch to control 2 relays, for some reason the sketch cannot set GPIO0 HIGH, GPIO2 is fine.  I know the ESP-01 is working because if I use a sketch without homie I can set GPIO0 HIGH.  Does homie  do something on GPIO0?

Thank you,

Mike

marvinroger, 2016-06-25:
See **Reset** : http://marvinroger.viewdocs.io/homie-esp8266~v1.5.0/3.-Advanced-usage

GPIO0 is indeed used as a trigger to reset the homie-esp8266 config. Before `Homie.setup()`, just do `Homie.disableResetTrigger()` to disable this behavior.

mikecay, 2016-06-25:
Thank you fir the quick response.  In doing so will I still be able to do ota updates?

Mike

marvinroger, 2016-06-26:
Yes you will. *Resetting* means, in Homie, press a button to erase the internal configuration of the board, OTA is another thing, and will stil work. :)

closed 2016-07-19 by marvinroger