homieiot/homie-esp8266 #106: Add method Homie.getNodeProperty

peterhanzlik, 2016-06-27:
It would be fine to be able to get the value of the desired property for the desired HomieNode:
`String Homie.getNodeProperty (HomieNode node, String property)`

Currently I need to maintain the last value of the property in a separate variable.

marvinroger, 2016-07-19:
The problem is we need to do dynamic allocation to implement this feature, which is bad for embedded systems. Currently, there are some heap allocations, but only at boot, and once, so there is no risk of heap fragmentation. However, dynamic allocation would be done each time `setNodeProperty` is called (continuously during the program flow) which would then be problematic on such a constrained environment.

Not every sketch needs this feature, and if you need this you only have to use an additional variable (preferably on the stack), is this really a big deal when you know this is way more safe in the long run?

peterhanzlik, 2016-07-19:
Hi Marvin,

I didn't know the background so just proposed this to make our codes neater.

Sure, if this new feature is dangerous for the Homie stability, then of course handling extra variable in my code is OK.

Thank you
Peter

marvinroger, 2016-08-12:
Closing it, the reason why is explained above. ESP8266 is I think too limited to implement this. Why not in a future `homie-esp32`, though. :angel:

closed 2016-08-12 by marvinroger