homieiot/homie-esp8266 #108: SSL

simpaltek, 2016-07-07:
Hi Marvin, 

I have been using your app since version 1.0 and I really like it.  However, your new version 2.0 does not handle SSL any more.  You stated that you removed this feature due to reliability issue.  Can you elaborate what you discovered about this?  Also, are you planning to put SSL back in future release?  If not, do you have any plans on making this the communication secure?

marvinroger, 2016-07-19:
SSL on Homie for ESP8266 [was a total mess](https://github.com/marvinroger/homie-esp8266/issues/46). The main problem is, with PubSubClient and SSL enabled, you cannot subscribe to 5 topics or more, otherwise it crashes. It was basically not stable at all, and firing 10 messages at it in a second would crash it. You agree this is not great for an IoT object to be this sensible. Finally, the SSL fingerprint checking implementation was, well, totally insecure, so it was pretty easy to do a MITM attack.

The 2.0 will use a new MQTT asynchronous library which is way, way more stable. Unfortunately, the underlying library does not support SSL ATM, but it is something we're working on.
So for now, communications will be in cleartext. This shoud not be too much of an issue in the context of a private Wi-Fi connection.

ioeverything, 2016-09-29:
I found Homie as a great work so i like its backend idea.
For a secure connection, we can encrypt payload and then send. SSL or TLS try to take time and resources on esp8266 so it cannot support it.
We can implement encrypt and decryption on callback and publish function to secure communication. It seems that it s more stable

marvinroger, 2016-10-07:
@ioeverything this would not prevent replay attack and would not allow authentication, it would only hide the payload. Which is the tiniest concern in our case.

averri, 2016-11-22:
I agree with @marvinroger. Security is something that you have or don't have, there is no such "partial security". I can't think about a IoT framework that does not support security. I understand the actual difficulty do add TLS, but we may thing about other ways to add it to Homie.

mkeyno, 2016-11-23:
@marvinroger  why don't you open homei gitter chat room to have better way to follow such important issue

marvinroger, 2016-11-30:
Now that [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP) supports SSL, there might actually be a way to make Homie for ESP8266 compatible with SSL/TLS again. I would not rely on it from a security standpoint, though.

I am reather busy these days, but SSL/TLS support is on the TO-DO list. 😉 

@mkeyno studying / keeping a social life / maintaining the project / answering on issues take time. I basically would have no time to follow a Gitter. But if that's a request, that might maybe be useful for the community (such a weird thing to tell myself Homie now has a community 😇 )

rohit-gandhe, 2017-02-09:
Any update on SSL? Homie is truly awesome, but can't really go to production without fixing this rather gaping hole. :)

mkeyno, 2017-02-09:
@marvinroger its truly promising lib specially written base on me-no-dev awesome async lib , why not you open the gitter and let your friends which have more time maintain it

n8twj, 2017-02-10:
https://gitter.im/homie-iot/ESP8266

marvinroger, 2017-02-10:
esp8266/Arduino 2.3.0 does not ship the needed code for the SSL part of me-no-dev/ESPAsyncTCP to work. So this won't be implemented until at least esp8266/Arduino 2.4.0 is released.

timpur, 2017-05-03:
Hi wondering how progress on mqtt ssl is going ? still waiting for arduino esp8266 2.4.0 ? Any one know when that will be ?

timpur, 2017-05-29:
Update 2.4.0 rc1 (https://github.com/esp8266/Arduino/releases/tag/2.4.0-rc1) has now been released :)
I know its still a while before homie will get ssl but ill be waiting for that moment :)

timpur, 2017-06-26:
@marvinroger Any updates on tls?
Ive had a look at async mqtt and it looks like tls is sported?
Is homie going to receive functionality to easily enable tls? or shall it be done through the async mqtt lib? if so any examples on how to do this which is compatible with homie? 

Tim.

svenove, 2018-04-08:
Any update on this? :)

timpur, 2018-04-08:
Has been added to V2.1

jmozmoz, 2018-05-13:
I tried using SSL/TLS with the current development version of homie. So my platformio.ini looks like:
```
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = esp01_1m
framework = arduino
build_flags = -DASYNC_TCP_SSL_ENABLED=1 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_SSL -DDEBUG_ESP_TLS_MEM 
  -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_UPDATE
lib_deps = https://github.com/marvinroger/homie-esp8266.git
```

And the config.json:
```
...
    "mqtt": {
      "host": "192.168.1.1",
      "port": 8883,
      "base_topic": "",
      "auth": false,
      "ssl": true,
      "ssl_fingerprint": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
    },
...
```
Unfortunately, it does not work. I had to add this to my main.cpp:
```
  Homie.setup();
#if ASYNC_TCP_SSL_ENABLED
  Homie.getLogger() << F("ESP Homie Using TLS") << endl;
  Homie.getMqttClient().setSecure(true);
#endif
```
If I remove setSecure(true), I cannot connect to the MQTT broker. The log file of it says:

    Client connection from 192.168.179.35 failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number.

With setSecure(true), it works.

jmozmoz, 2018-05-13:
The user interface for creating the config currently does not support switching on SSL and the input of the fingerprint

TuxCoder, 2018-07-25:
Hello all, I have a test implementation that works pretty well: https://github.com/TuxCoder/homie-esp8266
The tested program code is hosted here: https://git.o-g.at/esp8266/homie

The cert check is missing currently. :/
Have to figure out how to forward/implement it,
and read into the current stable version of the esp8266 toolchain.

The dev toolchain has the ability to check against a root cert. (used letsencrypt root)
I have already done that with my own framework, that I throw away for Homie.
Will try to give you a update in the next week(s).

timpur, 2018-07-25:
Sorry guys I'm still committed to homie just really busy ATM.

jmozmoz, 2018-07-25:
@TuxCoder: have you tested if OTA updates of the firmware work using SSL? I see these problems: https://github.com/marvinroger/homie-esp8266/issues/532

TuxCoder, 2018-07-25:
It also has worked, but failed sometimes, have to debug more.
Thought until yet, the problem is the small storage from the sonoff devices with 1MB.
But as the updates sometimes work sometimes not, it looks more likely the issue is TLS.

stritti, 2020-01-01:
No progress for long time. Closing it.

closed 2020-01-01 by stritti

kleini, 2020-01-01:
I would rather say, that this works again with develop-v3 branch. At least all my devices (5 currently) work with reliable SSL connections to the MQTT broker.

robertochajin, 2020-05-21:
I'm trying to connect to Mosquitto broker with Let's Encrypt SSL using Homie 3.0.0 and always is impossible.  This whats the Mosquitto logs print:

Client connection from 179.13.148.144 failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number.