homieiot/homie-esp8266 #109: use standard ArduinoOTA

lighthousebulb, 2016-07-10:
I do have a question:
Is it possible to use the old standard Arduino-OTA-Method somehow?

I have done a
`#include <ArduinoOTA.h>`
and use a 
`ArduinoOTA.handle();` inside a homie-LoopFunction,
but the node is not responding from within platform-io's OTA Upload-Method.
I looks like homie overwrites ArduinoOTA.

It would be very handy, during development to have the standard Arduino-OTA-Method from within platform-io.

Any suggestions?

lighthousebulb, 2016-07-10:
Awesome! Got it working...
Thanks for reading my question, but you can close this ;-)

You need to send something to the $ota-Topic first and then the node responses to a platform-io-OTA-request as well.

marvinroger, 2016-07-19:
Glad you got it working!

closed 2016-07-19 by marvinroger