homieiot/homie-esp8266 #111: Add device property with Homie version

kwo, 2016-07-15:
Is it worthwhile to add a device property with the Homie version? 

I see that the OTA mechanism will change in 2.0 from HTTP download to MQTT message. It would help OTA servers to know which Homie version a device is running to know how to make the firmware image available.

jpmens, 2016-07-15:
I think a property with Homie version number would be good to have. (What
I've done so far is to encode that into the firmware name...)

As far as the homie-ota server is concerned, what we'll probably do is to
either make firmware available via HTTP and MQTT or simply bite the bullet
and define that as from a particular moment we will no longer support HTTP.

>

kwo, 2016-07-15:
If you tag the homie-ota releases :-) we can run the current, 1.5-compatible version until all nodes have upgraded to the homie 2.0 before running a new version of homie-ota.

Nevertheless, it might still nice to know the Homie version of a device. Or perhaps be able to set arbitrary device properties in the same way that Homie.setNodeProperty allows us to set arbitrary node properties.

jpmens, 2016-07-15:
Touché! ;-)

>

marvinroger, 2016-08-16:
Done, a new device property `$homie` now exists. You can check the [Homie convention v2.0.0](https://github.com/marvinroger/homie/tree/v2) (well, it's still a draft, but you get the idea). ;)

@jpmens OTA and reset stuff is now implementation specific. (the `$ota/payload` works well for homie-esp8266 but it might not fit other implementations, so there are now implementation specific topics in Homie, see https://github.com/marvinroger/homie-esp8266/blob/master/docs/8.-Homie-implementation-specifics.md). It feels cleaner I think. :smiley:

marvinroger, 2016-08-16:
I said *done* but it's not actually released, I'll bump all v2.0.0 related issues when it is!

closed 2016-08-16 by marvinroger