homieiot/homie-esp8266 #112: Light controlled via mqtt and switch at same time

plosi, 2016-07-17:
Marvin, thank you very much for your great work. I have a noob question: I am trying to make a light controlled via mqtt (home-assistant.io) and a push button/switch at the same time (I am using a sonoff). The code below works alright, the only thing is that when I use the switch to turn on/off the light that was previously operated via the mqtt control, nothing happens and I need to push it again to make it work, and viceversa if I use the mqtt interface to operate the light previously turned on/off with the switch.
I am definitely making something wrong but cannot see what.
Thanks in advance for your help.

#include <Homie.h>

const int PIN_RELAY = 12;
const int PIN_LED = 13;
const int PIN_BUTTON = 0;

Bounce debouncerButton = Bounce();
int lastButtonValue = -1;

HomieNode switchNode("switch", "switch");

void buttonPressHandler(){
  int buttonValue = debouncerButton.read();
  int relayValue = digitalRead(PIN_RELAY);
  
  if (buttonValue != lastButtonValue){ //if button state has changed
    lastButtonValue = buttonValue;

    if (buttonValue == HIGH){ //only toggle relay if the new button state is HIGH
      digitalWrite(PIN_RELAY, !relayValue);
      Homie.setNodeProperty(switchNode, "on", relayValue ? "true" : "false", false);
    }
  }
}

bool switchOnHandler(String value) {
  if (value == "true") {
    digitalWrite(PIN_RELAY, HIGH);
    Homie.setNodeProperty(switchNode, "on", "true");
    Serial.println("Switch is on");
   } else if (value == "false") {
    digitalWrite(PIN_RELAY, LOW);
    Homie.setNodeProperty(switchNode, "on", "false");
    Serial.println("Switch is off");
   } else {
    return false;
   }
   return true;
}

void setup() {
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW);
  pinMode(PIN_BUTTON, INPUT);
  digitalWrite(PIN_BUTTON, HIGH);
  debouncerButton.attach(PIN_BUTTON);
  debouncerButton.interval(50);

  Homie.setFirmware("itead-sonoff", "1.0.0");
  Homie.setLedPin(PIN_LED, LOW);
  Homie.setResetTrigger(PIN_BUTTON, LOW, 10000);
  switchNode.subscribe("on", switchOnHandler);
  Homie.registerNode(switchNode);
  Homie.setLoopFunction(buttonPressHandler);
  Homie.setup();
}

void loop() {
  Homie.loop();
  debouncerButton.update();
}

marvinroger, 2016-07-19:
To be honest, I don't see anything wrong in this code... I'll test when I'll have time to. :)

plosi, 2016-07-19:
Thanks Marvin, that is much appreciated. I will keep debugging and will post an update if I can get it to work as expected.
Keep up the good work. This framework is awesome!

plosi, 2016-07-28:
Marvin,
after some debugging I have sorted things out: when the button is pressed, the latest value of variable holding relay state is reversed. The correct code looks like this:

```
if (buttonValue == HIGH){ //only toggle relay if the new button state is HIGH
      digitalWrite(PIN_RELAY, !relayValue);
      Homie.setNodeProperty(switchNode, "on", !relayValue ? "true" : "false");
    }
```
I still have one issue to fix: I was expecting the push button/switch to work even with no internet connection and/or without connection to the mqtt broker. However the node must be connected to the broker for the button to toggle the relay. Am I missing something?
Thanks

marvinroger, 2016-07-28:
@plosi Great!
The loop handler is only called when the connection to Wi-Fi/MQTT is up. But in your case you have a button so indeed, the loop handler is not the best place to do this.

Cut/paste the code from the loop handler to the main Arduino `loop()`, which is called every time.

TommySharpNZ, 2016-07-29:
@plosi can you post your completed code once you have it working nicely?

plosi, 2016-07-29:
I will for sure but I'm off on vacation and will return in a couple of weeks. Will post it as soon as I am back.

yves-ledermann, 2016-08-15:
@TommySharpNZ 
I just uploaded a version based on the work of @plosi 
On my version the button is also working offline.
https://github.com/yves-ledermann/arduino_esp8266_homie_sonoff
maybe there is a better aproach to do this... lets see how @plosi code looks..

plosi, 2016-08-19:
@TommySharpNZ  sorry for the delay. I just followed @marvinroger 's advice and moved the button function from the loop handler to the main Arduino loop(). The code updated looks as follows and it works nicely:
```
#include <Arduino.h>
#include <Homie.h>

const int PIN_RELAY = 12;
const int PIN_LED = 13;
const int PIN_BUTTON = 0;

Bounce debouncerButton = Bounce();
int lastButtonValue = -1;

HomieNode switchNode("switch", "switch");

#define FW_NAME "itead-sonoff"
#define FW_VERSION "1.0.1"

/* Magic sequence for Autodetectable Binary Upload */
const char *__FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" FW_NAME "\x93\x44\x6b\xa7\x75";
const char *__FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" FW_VERSION "\xb0\x30\x48\xd4\x1a";
/* End of magic sequence for Autodetectable Binary Upload */

void buttonPressHandler(){
  int buttonValue = debouncerButton.read();
  int relayValue = digitalRead(PIN_RELAY);

  if (buttonValue != lastButtonValue){ //if button state has changed
    lastButtonValue = buttonValue;

    if (buttonValue == HIGH){ //only toggle relay if the new button state is HIGH
      digitalWrite(PIN_RELAY, !relayValue);
      Homie.setNodeProperty(switchNode, "on", !relayValue ? "true" : "false", true);
    }
  }
}

bool switchOnHandler(String value) {
  if (value == "true") {
    digitalWrite(PIN_RELAY, HIGH);
    Homie.setNodeProperty(switchNode, "on", "true");
    Serial.println("Switch is on");
   } else if (value == "false") {
    digitalWrite(PIN_RELAY, LOW);
    Homie.setNodeProperty(switchNode, "on", "false");
    Serial.println("Switch is off");
   } else {
    return false;
   }
   return true;
}

void setup() {
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW); //relay is off when pin is low
  pinMode(PIN_BUTTON, INPUT);
  digitalWrite(PIN_BUTTON, HIGH);
  debouncerButton.attach(PIN_BUTTON);
  debouncerButton.interval(50);

  Homie.setFirmware(FW_NAME, FW_VERSION);
  Homie.setLedPin(PIN_LED, LOW);
  Homie.setResetTrigger(PIN_BUTTON, LOW, 5000);
  switchNode.subscribe("on", switchOnHandler);
  Homie.registerNode(switchNode);
  //Homie.setLoopFunction(buttonPressHandler);
  Homie.setup();
}

void loop() {
  Homie.loop();
  buttonPressHandler();
  debouncerButton.update();
}
```

the solution proposed  by @yves-ledermann is very neat and nice indeed, I might give it a try. In the meantime, I would like to install the sonoff within a standard electrical wall box and see how it behaves. My only concern would be if it get too warm.

@marvinroger: I consider the issue/question resolved. Please feel free to tag it as closed or leave it open for further discussion on the matter. Thank you very much!

marvinroger, 2016-08-19:
Awesome, thanks! Let's close this issue. Feel free to continue the discussion if you wish to, but this is not an actual *issue* anymore. :wink:

closed 2016-08-19 by marvinroger