homieiot/homie-esp8266 #113: Exception on going to deep sleep

DaSepp, 2016-07-20:
On my sensor node i want to fall into deep sleep for a specific time span.

I am doing this: 

```
void loopHandler() {
	//some code before

	Homie.disconnectMQTT();
	delay(100);
}
```

After MQTT has disconnected i´m catchig the event to fall into deepsleep.

```
void onHomieEvent(HomieEvent event) {
  switch(event) {
    case HOMIE_MQTT_DISCONNECTED:
      // Do whatever you want when MQTT is disconnected in normal mode
      ESP.deepSleep(5000000);
      delay(100);
      break;
  }
}
```

On reaching the deep sleep command all i got is a exception.

Has anybody undertaken same issue?

marvinroger, 2016-07-21:
Please try something like this:

```c++
bool sleepFlag = false;

void onHomieEvent(HomieEvent event) {
  switch(event) {
    case HOMIE_MQTT_DISCONNECTED:
      sleepFlag = true;
      break;
  }
}

void loop() { // The main Arduino loop, not the loopHandler
  if (sleepFlag == true) {
    ESP.deepSleep(5000000);
  }
}
```

marvinroger, 2016-08-12:
Just pinging you @DaSepp, any news? Otherwise I'll close this, as this is moreover an esp8266/Arduino issue more than an issue with homie-esp8266.

DaSepp, 2016-08-12:
As i´m very busy the last weeks i didn´t verify your solution. But anyway, as nobody else is effected you can close this thread.

closed 2016-08-12 by DaSepp

marvinroger, 2016-08-12:
No problem, feel free to reopen if needed. :)