homieiot/homie-esp8266 #116: .subscribe() does not register

lighthousebulb, 2016-07-28:
This got me pretty frustrated yesterday. But it seems that none of my nodes will register properly. The `.subscribe()`-function does not work. This is really wired, because I wrote sketches before and it just worked. Now even the example stopped working.

I used a up to date platform-io environment and even tried it with a up to date Adruino-IDE.
After some frustrating hours I tried the LightOnOff.ino-Example.

The homie-node comes online and sends it initial Device properties. But the $nodes-property keeps empty. If I try to set the light „on“ I just get something like “Node xxx not registered” on the serial line.

marvinroger, 2016-07-28:
Are you using the v1.5.0?

lighthousebulb, 2016-07-28:
Yes! v1.5.0

marvinroger, 2016-07-28:
So you say it was working before... What have you done in this timelapse? Maybe you've updated the esp8266/Arduino version? Which version by the way, 2.2.0, 2.3.0?

peterhanzlik, 2016-07-28:
I had the same problem until I noticed that I was missing: `Homie.registerNode(lightNode);`. In v1.5.0 it is mandatory, in v2.0.0 it is not used. I mixed up examples from v2 and tried to use them in v1.5. Perhaps the same happened in your case as well.

marvinroger, 2016-07-28:
Oh yes, this might be a problem. You have to use the docs and examples contained in your release, not in the Git development version (unless you want to test cutting edge features, not stable... at all!).

lighthousebulb, 2016-07-28:
Awesome! Yes, that was the problem!

closed 2016-07-28 by lighthousebulb