homieiot/homie-esp8266 #117: Inability to revert to configuration mode on Wemos D1 mini

wolass, 2016-07-28:
Hi,

I did configuration step using API as indicated in the docs. I'm using the latest stable release. 

The problem with Wemos mini board is that there is no additional button for "flashing". Only reset. 

This is why I cannot get the board into configuration again. 

I tried reflahing with other arduino sketches but after uploading basic homie, the scripts gets the previous config.jason file from somewhere. And I do not think that it is dokumented from where.

1. Please advise on how to get back to configuration on boards without a button.
2. Please make it clearer in the docs how to change config.json manually not using the API. 

Thanks.
W

marvinroger, 2016-07-28:
You can change the `FLASH` button with another pin, or set your own reset function: http://marvinroger.viewdocs.io/homie-esp8266~v1.5.0/3.-Advanced-usage/#user-content-reset

To change `config.json` manually: http://marvinroger.viewdocs.io/homie-esp8266~v1.5.0/5.-JSON-configuration-file/

> To configure your device, you have two choices: manually flashing the configuration file to the SPIFFS at the /homie/config.json (see [Uploading files to file system](http://esp8266.github.io/Arduino/versions/2.1.0/doc/filesystem.html#uploading-files-to-file-system)), so you can bypass the configuration mode, or send it through the Configuration API.

**tl;dr** 

Put your `config.json` in a `data/homie/config.json` in the directory of your sketch, and select `Tools > ESP8266 Sketch Data Upload`

If you think docs are not clear enough, please open a PR with the changes you'd like. :)

NigelTwo, 2016-08-09:
Wolass, the SPIFFS file system is written into a different "bank" of the ESP8266 flash memory from what is used for sketch program space. That is why it was persistent after you programmed a different sketch onto the ESP8266 module.
I have struggled to get a syntatically correct config.json file! So just create a config.json file with some random text and upload it. Take care to observe the full path (/homie/config.data) of the image that SPIFFS is creating for you.

wolass, 2016-08-09:
Thx NigelTwo and Marvin. Actually after marvin gave the link  http://marvinroger.viewdocs.io/homie-esp8266~v1.5.0/5.-JSON-configuration-file/ it all became clear! But if someone is not aware that ESP8266 has internal memory and you need a specific TOOL for the arduino IDE to be able to write in this memory (e.g. me :) ) - its not that easy to figure that on your own from the docs. I'll try to suggest a PR, but now I'm swamped with work...