homieiot/homie-esp8266 #118: OTA during development

lighthousebulb, 2016-07-28:
Some time ago I experimented with the ArduinoOTA-feature. This is a very nice feature during development, because flashing a modified sketch is much faster OTA, than over serial. 
Now in homie there is a different OTA approach. This is fine in a production environment.

Is there a way to use the built in OTA-flash-function in the Adruino IDE or Platform-IO with homie?
I mean just hit upload and it flashes the current sketch to a specific IP-Address?

How do you guys do this in your development environment?

marvinroger, 2016-07-28:
You can still use the ArduinoOTA code in your sketch if you want to, I have not tested but it should work.

However, I won't provide directions for OTA over Homie for v1.x.x as the OTA procedure for the v2 is completely different (over MQTT directly). You'll simply have to send the firmware on a specific topic, and the device will start flashing it. This should be doable directly on the Arduino IDE, with a plugin and some `mosquitto_pub` binaries. I will take a look at it later, when the v2 is released (no ETA, unfortunately).

yves-ledermann, 2016-08-15:
Hi @lighthousebulb 
I just tried the arduinoOTA according a sketch of  @unaiur 
https://github.com/unaiur/homie-nodes/tree/master/examples/ledstrip/src

You find my version on https://github.com/yves-ledermann/arduino_esp8266_homie_sonoff
Its not much tested but it seems to work.
Sometimes i had to do reapply power after update... i need to take a closer look...

shogsbro, 2016-08-31:
+1 for easy OTA upload during development. Could be made a compile-time switch, so that release build would be a smaller binary image.

closed 2020-01-04 by stritti