homieiot/homie-esp8266 #119: Need example of subscribe at Homie.

sundariitm, 2016-07-30:
homie-esp8266-1.5.0\examples\LightOnOff

I am trying to run the example as-is from this folder and publish data from a mqtt server. However i am seeing the Serial Monitor console are not showing any prints.

I tried publishing "false" or "true" to the following nodes but none of them was seen in serial console.

devices/devid/light/switch
devices/devid/light/switch/on
devices/devid/lightNode/switch
devices/devid/lightNode/switch/on

devices & devid are the values i use in homie configurator web interface 

Only following is seen in console
Triggering HOMIE_MQTT_CONNECTED event...
Calling setup function...
Sending Wi-Fi signal quality (52%)...  OK
Sending uptime (6s)...  OK
Sending uptime (126s)...  OK
Sending uptime (246s)...  OK
Sending Wi-Fi signal quality (52%)...  OK
Sending uptime (366s)...  OK
Sending uptime (486s)...  OK
Sending Wi-Fi signal quality (48%)...  OK
Sending uptime (606s)...  OK
Sending uptime (726s)...  OK

lighthousebulb, 2016-07-30:
You might want to have a look at the issue #116

sundariitm, 2016-07-30:
marvin

i looked at the thread. I checked all the points there and confirmed i am
not caught in those

My doubt here is on the publish topic hierarchy on which the lightOnOff
example (in 1.5 release) is subscribing to?

From the MQTT client what topic hierarchy must be published with true/false
so that the light to go ON/OFF and see the prints in console?

thx
sundar

marvinroger, 2016-07-31:
`devices/devid/light/switch/on/set` :)
Read the Homie convention, you'll have a better understanding.

marvinroger, 2016-07-31:
`devices/devid/light/on/set`, my bad

sundariitm, 2016-08-01:
thanks for pointing out. i was always searching within homie-esp8266 folder
for convention :)

marvinroger, 2016-08-02:
You're welcome.

closed 2016-08-02 by marvinroger

valentinvieriu, 2016-08-05:
@sundariitm did you managed to make the LightOnOff example work?
I'm facing a similar situation now, and getting the error
`Node light not registered`

sundariitm, 2016-08-06:
Are you using the example as is?  pay attention to versions, do not mix examples and homie version
Or check if you have this statement in your program Homie.registerNode(lightNode); 
~sundariitm

valentinvieriu, 2016-08-06:
Thank you @sundariitm . The mix in version was the problem. :)