homieiot/homie-esp8266 #120: https://github.com/me-no-dev/ESPAsyncWebServer

kiralikbeyin, 2016-07-31:
Nice work.

Can i use https://github.com/me-no-dev/ESPAsyncWebServer?

Any example servering its own files on nodemcu?

Why dont you use mosca for server?

Thanks.

marvinroger, 2016-07-31:
Thanks!

You can use ESPAsyncWebServer or any libraries, as usual... The v2 will actually use this library internally. You can find examples on the ESPAsyncWebServer repo, this is not related to this project.

Well, you can also use mosca, mosquitto, any MQTT broker you want to.

kiralikbeyin, 2016-07-31:
When i add these libs to one of your examples i got errors. 

#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>

Is this happening to you?

marvinroger, 2016-07-31:
What are the errors?

kiralikbeyin, 2016-07-31:
```
In file included from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:3:0:
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:55:3: error: redeclaration of 'HTTP_ANY'
   HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_DELETE, HTTP_PUT, HTTP_PATCH, HTTP_HEAD, HTTP_OPTIONS
   ^
In file included from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootConfig.hpp:7:0,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.hpp:12,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.h:4,
                 from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:1:
/Users/EvAkilli/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:30:19: note: previous declaration 'HTTPMethod HTTP_ANY'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                   ^
In file included from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:3:0:
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:55:13: error: redeclaration of 'HTTP_GET'
   HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_DELETE, HTTP_PUT, HTTP_PATCH, HTTP_HEAD, HTTP_OPTIONS
             ^
In file included from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootConfig.hpp:7:0,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.hpp:12,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.h:4,
                 from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:1:
/Users/EvAkilli/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:30:29: note: previous declaration 'HTTPMethod HTTP_GET'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                             ^
In file included from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:3:0:
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:55:23: error: redeclaration of 'HTTP_POST'
   HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_DELETE, HTTP_PUT, HTTP_PATCH, HTTP_HEAD, HTTP_OPTIONS
                       ^
In file included from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootConfig.hpp:7:0,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.hpp:12,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.h:4,
                 from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:1:
/Users/EvAkilli/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:30:39: note: previous declaration 'HTTPMethod HTTP_POST'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
In file included from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:3:0:
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:55:34: error: redeclaration of 'HTTP_DELETE'
   HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_DELETE, HTTP_PUT, HTTP_PATCH, HTTP_HEAD, HTTP_OPTIONS
                                  ^
In file included from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootConfig.hpp:7:0,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.hpp:12,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.h:4,
                 from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:1:
/Users/EvAkilli/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:30:72: note: previous declaration 'HTTPMethod HTTP_DELETE'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                                                        ^
In file included from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:3:0:
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:55:47: error: redeclaration of 'HTTP_PUT'
   HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_DELETE, HTTP_PUT, HTTP_PATCH, HTTP_HEAD, HTTP_OPTIONS
                                               ^
In file included from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie/Boot/BootConfig.hpp:7:0,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.hpp:12,
                 from /Users/EvAkilli/Documents/Arduino/libraries/homie-esp8266-master/src/Homie.h:4,
                 from /var/folders/my/_g1m35d57k9353s6nkyym9d40000gn/T/arduino_modified_sketch_129386/HookToEvents.ino:1:
[... cut ...]
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlerImpl.h: In member function 'virtual bool AsyncCallbackWebHandler::canHandle(AsyncWebServerRequest*)':
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlerImpl.h:78:21: warning: comparison between 'enum WebRequestMethod' and 'enum HTTPMethod' [-Wenum-compare]
       if(_method != HTTP_ANY && request->method() != _method)
                     ^
Multiple libraries were found for "Bounce2.h"
 Used: /Users/EvAkilli/Documents/Arduino/libraries/Bounce2
 Not used: /Users/EvAkilli/Documents/Arduino/libraries/Bounce2-master
```

kiralikbeyin, 2016-07-31:
Sketch not changed only added 2 libs..

```
#include <Homie.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
void onHomieEvent(HomieEvent event) {
  switch(event) {
    case HOMIE_CONFIGURATION_MODE:
      Serial.println("Configuration mode started");
      break;
    case HOMIE_NORMAL_MODE:
      Serial.println("Normal mode started");
      break;
    case HOMIE_OTA_STARTED:
      Serial.println("OTA started");
      break;
    case HOMIE_OTA_FAILED:
      Serial.println("OTA failed");
      break;
    case HOMIE_OTA_SUCCESSFUL:
      Serial.println("OTA successful");
      break;
    case HOMIE_ABOUT_TO_RESET:
      Serial.println("About to reset");
      break;
    case HOMIE_WIFI_CONNECTED:
      Serial.println("Wi-Fi connected");
      break;
    case HOMIE_WIFI_DISCONNECTED:
      Serial.println("Wi-Fi disconnected");
      break;
    case HOMIE_MQTT_CONNECTED:
      Serial.println("MQTT connected");
      break;
    case HOMIE_MQTT_DISCONNECTED:
      Serial.println("MQTT disconnected");
      break;
  }
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Homie.enableLogging(false);
  Homie_setFirmware("events-test", "1.0.0");
  Homie.onEvent(onHomieEvent);
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

marvinroger, 2016-07-31:
Oh yes, the "native" WebServer is used internally, so it conflicts with the 3rd party async WebServer. Unfortunately, it won't work until v2 is released.

ntalekt, 2017-05-16:
I'm having this same issue too. What was wrong with the travis build? The logs aren't available.

closed 2019-03-23 by stritti