homieiot/homie-esp8266 #121: Cannot subscribe to topic - LightOnOff not working

valentinvieriu, 2016-08-05:
Hi,
I'm trying to implement something very similar to LightOnOff example. I've tried first with the LightOnOff example out of the box, with the version 1.5 of homie-esp8266.

I've tried your advice from #119 and still cannot make it work.
My deviceid is esp01. I've loaded the example file and then hitting this endpoint with a MQTT client
`devices/esp01/light/on/set` and I'm getting 
`Node light not registered`

Any help is appreciated

marvinroger, 2016-08-05:
Hi! 

Can you please paste the output of the Serial monitor on boot? (showing the configuration)

valentinvieriu, 2016-08-05:
Sure, 
Here it is.

```
 1384, room 16 
tail 8
chksum 
��W���d
v3de0c112
~ld

Triggering HOMIE_NORMAL_MODE event...
** Booting into normal mode **
{} Stored configuration:
  • Hardware device ID: d233bee0
  • Device ID: esp01
  ��� Boot mode: normal
  • Name: esp 01
  • Wi-Fi
    ◦ SSID: surpriza
    ◦ Password not shown
  • MQTT
    ◦ Host: 192.168.1.120
    ◦ Port: 1883
    ◦ Base topic: devices/
    ◦ Auth? no
    ◦ SSL? no
  • OTA
    ◦ Enabled? yes
    ◦ Host: 192.168.1.120
    ◦ Port: 9080
    ��� Path: /ota
    ◦ SSL? no
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected
Triggering HOMIE_WIFI_CONNECTED event...
��� Attempting to connect to MQTT...
Connected
Sending initial information...  OK
Subscribing to topics...  OK
✔ MQTT ready
Triggering HOMIE_MQTT_CONNECTED event...
Calling setup function...
Sending Wi-Fi signal quality (66%)...  OK
Sending uptime (35s)...  OK
```

Please let me know if there is anything I can provide. The error message apears only when I send the request to `devices/esp01/light/on/set` from another MQTT client. 
Pushing works very well. I can send data from the ESP to MQTT. I've  tried already with DHT 22 sensor.

Thank you for the assistance!

valentinvieriu, 2016-08-05:
Here is a look at the logs from the MQTT server
```
mosquitto_1  | 1470430965: Received SUBSCRIBE from Homie-esp01                                                                                                                      
mosquitto_1  | 1470430965:      devices/esp01/+/+/set (QoS 1)                                                                                                                       
mosquitto_1  | 1470430965: Homie-esp01 1 devices/esp01/+/+/set                                                                                                                      
mosquitto_1  | 1470430965: Sending SUBACK to Homie-esp01                                                                                                                            
mosquitto_1  | 1470430965: Received SUBSCRIBE from Homie-esp01                                                                                                                      
mosquitto_1  | 1470430965:      devices/esp01/$reset (QoS 1)                                                                                                                        
mosquitto_1  | 1470430965: Homie-esp01 1 devices/esp01/$reset                                                                                                                       
mosquitto_1  | 1470430965: Sending SUBACK to Homie-esp01                                                                                                                            
mosquitto_1  | 1470430965: Received SUBSCRIBE from Homie-esp01                                                                                                                      
mosquitto_1  | 1470430965:      devices/esp01/$ota (QoS 1)                                                                                                                          
mosquitto_1  | 1470430965: Homie-esp01 1 devices/esp01/$ota                                                                                                                         
mosquitto_1  | 1470430965: Sending SUBACK to Homie-esp01                                                                                                                            
mosquitto_1  | 1470430965: Received PUBLISH from Homie-esp01 (d0, q0, r1, m0, 'devices/esp01/$signal', ... (2 bytes))                                                               
mosquitto_1  | 1470430965: Sending PUBLISH to homie-ota-6 (d0, q0, r0, m0, 'devices/esp01/$signal', ... (2 bytes))                                                                  
mosquitto_1  | 1470430965: Sending PUBLISH to 9e5a410b2fcf412dadbc44108650ac54 (d0, q0, r0, m0, 'devices/esp01/$signal', ... (2 bytes))                                             
mosquitto_1  | 1470430965: Received PUBLISH from Homie-esp01 (d0, q0, r1, m0, 'devices/esp01/$uptime', ... (1 bytes))                                                               
mosquitto_1  | 1470430965: Sending PUBLISH to homie-ota-6 (d0, q0, r0, m0, 'devices/esp01/$uptime', ... (1 bytes))                                                                  
mosquitto_1  | 1470430965: Sending PUBLISH to 9e5a410b2fcf412dadbc44108650ac54 (d0, q0, r0, m0, 'devices/esp01/$uptime', ... (1 bytes))                                             
mosquitto_1  | 1470430976: Received PINGREQ from homie-ota-6                                                                                                                        
mosquitto_1  | 1470430976: Sending PINGRESP to homie-ota-6                                                                                                                          
mosquitto_1  | 1470430980: Received PINGREQ from Homie-esp01                                                                                                                        
mosquitto_1  | 1470430980: Sending PINGRESP to Homie-esp01                                                                                                                          
mosquitto_1  | 1470430995: Received PINGREQ from Homie-esp01                                                                                                                        
mosquitto_1  | 1470430995: Sending PINGRESP to Homie-esp01                                                                                                                          
mosquitto_1  | 1470431010: Received PINGREQ from Homie-esp01                                                                                                                        
mosquitto_1  | 1470431010: Sending PINGRESP to Homie-esp01                                                                                                                          
mosquitto_1  | 1470431016: Received PINGREQ from 9e5a410b2fcf412dadbc44108650ac54                                                                                                   
mosquitto_1  | 1470431016: Sending PINGRESP to 9e5a410b2fcf412dadbc44108650ac54                                                                                                     
mosquitto_1  | 1470431025: Received PINGREQ from Homie-esp01                                                                                                                        
mosquitto_1  | 1470431025: Sending PINGRESP to Homie-esp01                                                                                                                          
[... cut ...] 0, q0, r0, m0, 'devices/esp01/light/on/set', ... (4 bytes))                                     
mosquitto_1  | 1470431025: Sending PUBLISH to 9e5a410b2fcf412dadbc44108650ac54 (d0, q0, r0, m0, 'devices/esp01/light/on/set', ... (4 bytes))                                        
mosquitto_1  | 1470431025: Sending PUBLISH to Homie-esp01 (d0, q0, r0, m0, 'devices/esp01/light/on/set', ... (4 bytes))   
```

So the device subscribe to a wildcard like subscription `devices/esp01/+/+/set`

Hope it helps

marvinroger, 2016-08-05:
Use the examples that are bundled with your release. v1.5.0 needs manual registering of the nodes, whereas this is automatic in the in-development v2.0.0, so current examples in the repo do not work with the latest release. :)

valentinvieriu, 2016-08-06:
👍  thank you. You have no idea how much time I've spent yesterday trying to figure this out.
Used the 1.5 example and it worked. As I've just started building this project ( I want to install network of sensors in all our meeting rooms at work ), would you suggest to use the 1.5 version, or should I already start with the 2.0?

Again, thank you for the quick reply and support.

closed 2016-08-06 by valentinvieriu

marvinroger, 2016-08-06:
You're welcome!

The v1.5 is pretty stable, and the v2.0 is far from finished. I don't know when I will be able mark it as stable, but it won't be that soon, so stick with v1.5 for the moment.