homieiot/homie-esp8266 #122: Unable to resolve DNS name of mqtt broker: exception and reset

clough42, 2016-08-08:
I am unable to use the DNS name of my MQTT broker.  It works fine if I provide the IP address.

This is using the latest code in the git repository.  If you would prefer not to have defect reports for the unreleased code, I will withdraw this issue.

I have tried both configuring with the captive portal and by uploading a config file.  On my network, the name "mqtt" resolves through DNS to the address of my broker--either as "mqtt" or "mqtt.clough.local".  I have tried both.  In each case, the WiFi connects, I get the message saying it's attempting to connect to the broker and then it exceptions and resets.  If I configure with the IP address of the broker, it works as expected.

I am using the latest code from the git repo (pre-2.0) and allowing pio to automatically resolve its dependencies, including async-mqtt-client.  I'm using this on a Wemos D1 Mini V2 with the latest version of Atom/PlatformIO.

clough42, 2016-08-11:
Is this unique to me?  Are other people successfully using hostnames?

atschuni, 2016-08-11:
same with me.

DavidStacer, 2016-08-11:
I had the same problem. I have a local DNS server (and active directory) where I was using home.local as the dns domain name.  The mqtt.home.local could not be resolved.  I created a new entry mqtt.junk.com in dns and it worked.  .local is reserved for multicast resolution.  History of .local in https://en.wikipedia.org/wiki/.local .

marvinroger, 2016-08-12:
I've been using [CloudMQTT](https://www.cloudmqtt.com/) without issues, so DNS resolution should be working on the v1.x. If there is an exception, it's probably an issue with esp8266/Arduino itself. Can you try with the esp8266/Arduino examples without Homie to see if I'm right?

clough42, 2016-08-23:
Thanks for the comment.  I still haven't gotten back around to testing this.  I've got so many projects going.  I will.

clough42, 2016-08-24:
I defined a new DNS entry with a valid TLD, and it still didn't work for me, but I haven't had time to dig into it further.  I'll report back once I do.

clough42, 2016-08-27:
I finally had a chance to test this weekend.  Here are the results:

- With Homie-esp8266 (2.0, tip of master)
-- 10.98.76.246:  Works
-- mqtt.clough.local:  Exception and reset
-- mqtt.home.clough42.com:  Exception and reset
-- m12.cloudmqtt.com:  Exception and reset

- With straight arduino sketch (From https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide)
-- www.adafruit.com:  Works

marvinroger, 2016-08-27:
Can you please update to the latest https://github.com/me-no-dev/ESPAsyncTCP ?

clough42, 2016-08-28:
Yes.  I pulled that down and linked it into my lib directory (I'm using PlatformIO) and now it works.  Thanks!

closed 2016-08-28 by clough42

uyounis, 2018-05-08:
Hi Guys, sorry to open this up gain.  Could one of you guys help me with this issue? I cannot use a hostname in my arduino sketch (ESP32) for the mqtt broker on my pi.  Does anyone have an exmaple I could use please?

Thanks

timpur, 2018-05-08:
What's your set up for Esp32 it can't be homie ? What code are you running ?

uyounis, 2018-05-08:
``#include <Arduino.h>
#include "DHT.h"
#include <PubSubClient.h>
#include <WiFi.h>
#include <Adafruit_Sensor.h>
#include <DHT.h>

// Uncomment one of the lines bellow for whatever DHT sensor type you're using!
#define DHTTYPE DHT11   // DHT 11
//#define DHTTYPE DHT21   // DHT 21 (AM2301)
//#define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321

// Change the credentials below, so your ESP8266 connects to your router
const char* ssid = "******";
const char* password = "*****";

//MQTT settings

#define MQTTid "1"
#define MQTTuser "****"
#define MQTTpsw "****"

// Change the variable to your Raspberry Pi IP address, so it connects to your MQTT broker
const char* mqttServer = "brew-malt-server1";

// Initializes the espClient. You should change the espClient name if you have multiple ESPs running in your home automation system
WiFiClient espClient;
PubSubClient client(espClient, mqttServer);

// DHT Sensor - GPIO 5 = D1 on ESP-12E NodeMCU board
const int DHTPin = 5;

// Lamp - LED - GPIO 4 = D2 on ESP-12E NodeMCU board
const int lamp = 4;

// Initialize DHT sensor.
DHT dht(DHTPin, DHTTYPE);

// Timers auxiliar variables
long now = millis();
long lastMeasure = 0;

// Don't change the function below. This functions connects your ESP8266 to your router
void setup_wifi() {
  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("WiFi connected - ESP IP address: ");
  Serial.println(WiFi.localIP());
}

// This functions is executed when some device publishes a message to a topic that your ESP8266 is subscribed to
// Change the function below to add logic to your program, so when a device publishes a message to a topic that 
// your ESP8266 is subscribed you can actually do something
void callback(String topic, byte* message, unsigned int length) {
  Serial.print("Message arrived on topic: ");
  Serial.print(topic);
  Serial.print(". Message: ");
  String messageTemp;
  
  for (int i = 0; i < length; i++) {
    Serial.print((char)message[i]);
[... 103 lines cut ...]
    // Uncomment to compute temperature values in Fahrenheit 
    // float hif = dht.computeHeatIndex(f, h);
    // static char temperatureTemp[7];
    // dtostrf(hic, 6, 2, temperatureTemp);
    
    static char humidityTemp[7];
    dtostrf(h, 6, 2, humidityTemp);

    // Publishes Temperature and Humidity values
    client.publish("room1/temperature", temperatureTemp);
    client.publish("room1/humidity", humidityTemp);
    
    Serial.print("Humidity: ");
    Serial.print(h);
    Serial.print(" %\t Temperature: ");
    Serial.print(t);
    Serial.print(" *C ");
    Serial.print(f);
    Serial.print(" *F\t Heat index: ");
    Serial.print(hic);
    Serial.println(" *C ");
    // Serial.print(hif);
    // Serial.println(" *F");
  }
} ``

uyounis, 2018-05-08:
Sorry for the unclear formatting.. This is my first post.

uyounis, 2018-05-08:
I am happy to utilise a esp8266 if I can connect using a hostname.  I am in a university campus network and do not have access to the router.

timpur, 2018-05-08:
That sketch has nothing to do with homie... we cant support. All i can say  is look at this https://github.com/knolleary/pubsubclient/pull/261. Please take your issue to relevant repo. 

Also Homie will soon come to esp32, but not yet, https://github.com/marvinroger/homie-esp8266/issues/415

uyounis, 2018-05-08:
Thank you for your repsonse, much apreciated.  I take it homie is a firmware build.  If I use this build on an ESP8266 will it enable me to use a hostname as a mqtt broker address instead of an static IP.

timpur, 2018-05-08:
Yes, but you can just the the Async mqtt lib homie is built on and should do the same, if you don't need homie.