homieiot/homie-esp8266 #124: Question about naming convention of nodes

claybar, 2016-08-09:
Hi,

I have a question about naming conventions of ids and properties for a 'sensor node'.  In the 'getting started' example there are the calls:
```
HomieNode temperatureNode("temperature", "temperature");
Homie.setNodeProperty(temperatureNode, "temperature", String(temperature), true);
```
Here a node is set up with a `id="temperature"` and `type="temperature"`.  Messages are then sent via the second call with `property="temperature"`.  Is this property a reference to the id of the first call, or can it be any string?

For my current application I am building a solar hot water cylinder monitor.  There are 4 temperature sensors, three mounted on the hot water cylinder (top, middle, bottom) and a 4th sensor on the solar collector on the roof.  So, for this setup am I best have 4 nodes, or 2 nodes with one of the nodes having 3 properties.  For the 4 node case the code would be along the lines of:
```
HomieNode cylinderTopNode("temperatureTop", "temperature");
HomieNode cylinderMidNode("temperatureMid", "temperature");
HomieNode cylinderBotNode("temperatureBot", "temperature");
HomieNode collectorNode("temperatureCollector", "temperature");
...
Homie.setNodeProperty(cylinderTopNode, "temperature", value, true);
Homie.setNodeProperty(cylinderMidNode, "temperature", value, true);
Homie.setNodeProperty(cylinderBotNode, "temperature", value, true);
Homie.setNodeProperty(collectorNode, "temperature", value, true);
```

Under the second scheme it would be more like:
```
HomieNode cylinderNode("temperatureCylinder", "temperature");
HomieNode collectorNode("temperatureCollector", "temperature");
...
Homie.setNodeProperty(cylinderNode, "temperatureTop", value, true);
Homie.setNodeProperty(cylinderNode, "temperatureMiddle", value, true);
Homie.setNodeProperty(cylinderNode, "temperatureBottom", value, true);
Homie.setNodeProperty(collectorNode, "temperature", value, true);
```

Or perhaps something different again would be better?

furyfire, 2016-08-12:
As I would see it you need to use the first scheme.
The "type" field is the representation of the value. The frontend should know from that field alone which properties are available to display. Alternatively you should make a 
`HomieNode cylinderNode("temperatureCylinder", "temperatureTMB");`
Which contains a Top, middle, bottom temperature so that given the type field temperatureTMB the frontend knows to display 3 temperatures in the data widget.

Just my 2 cents.

marvinroger, 2016-08-12:
```cpp
HomieNode temperatureNode("temperature", "temperature");
Homie.setNodeProperty(temperatureNode, "temperature", String(temperature), true);
```

From [the Homie convention](https://github.com/marvinroger/homie):

> An instance of a physical piece of hardware (an Arduino, an ESP8266...) is called a **device**. A device has **device properties**, like the current local IP, the Wi-Fi signal, etc. A device can expose multiple **nodes**. For example, a weather device might expose a temperature node and an humidity node. A node can have multiple **node properties**. The temperature node might for example expose a temperature property containing the actual temperature, and an unit property.

So no, the `setNodeProperty` `"temperature"` is not a reference to the id of the node. Maybe it would be cleaner if written this way:

```cpp
HomieNode temperatureNode("basementtemperature", "temperature");
Homie.setNodeProperty(temperatureNode, "unit", "c", true);
Homie.setNodeProperty(temperatureNode, "degrees", String(temperature), true);
```

About your second question: do what you think works best for your setup. I don't want to "enforce" a way of doing things, it all depends. @furyfire has its own, and it works well (that's how I would have done it too):

```cpp
HomieNode cylinderNode("temperatureCylinder", "temperatureTMB");
HomieNode collectorNode("temperatureCollector", "temperature");
...
Homie.setNodeProperty(cylinderNode, "temperatureTop", value, true);
Homie.setNodeProperty(cylinderNode, "temperatureMiddle", value, true);
Homie.setNodeProperty(cylinderNode, "temperatureBottom", value, true);
Homie.setNodeProperty(collectorNode, "temperature", value, true);
```

This is best because this way you consider things as logical `entities`. The 3 temperature sensors on the cylinder are actually part of the cylinder, and the one on the roof is independant. So yes, two nodes are best suited. But again, this is subjective.

euphi, 2016-08-12:
I would prefer the second variant, because it requires fewer node instances.
With Homie 2.0 there is no more hard limit of nodes, but it is still some unecessary overhead.

Also, when you think about the "data model" , the Cylinder is one item with several properties.

Regarding the `ID` and `type` of a Homie node: As far as I know there is no convention yet, except that they should somehow describe the type of the Node. Maybe its a good idea to set the type to "cylinder" instead of "temperature".

BTW: Questions about convention should go to the Homie convention project: https://github.com/marvinroger/homie

marvinroger, 2016-08-12:
@euphi you're right, `cylinder` as the type would be even better!

claybar, 2016-08-14:
Thanks for the guidance, the clarification of nodes and properties helps.  I think I will implement things as:
```
HomieNode cylinderNode("hotWaterCylinder", "hwcylinder");
HomieNode collectorNode("solarCollector", "solarcollector");
...
Homie.setNodeProperty(cylinderNode, "temperatures", "top,mid,bot", true);
Homie.setNodeProperty(collectorNode, "temperature", value, true);
```
My plan is to channel the messages into InfluxDB for logging and graphing via Graphite, so hopefully I can find a way to isolate the three temperature values for graphing.  The added bonus of this scheme is the three measurements can have equal timestamps in the database rather than a few ms different. 

This scheme should also allow me to add another property to the cylindernode for the electric heating element if I can figure out hardware to source the info:
```
Homie.setNodeProperty(cylinderNode, "element", on|off, true);
```
Likewise for the pump controlling the collector:
```
Homie.setNodeProperty(collectorNode, "pump", on|off, true);
```

closed 2016-08-14 by claybar

flaviostutz, 2016-08-15:
I am doing the same using InfluxDB, how are you getting data from mqtt and posting on InfluxDB? A self made program?