homieiot/homie-esp8266 #129: PlatformIO Library install is missing PubSubClient library dependency

javamatte, 2016-08-17:
I have PlatformIO CLI only (no ide, using SublimeText) and installed the Homie library using the following command:  (output below)
```
platformio lib install 555 
```
That worked, but trying to 'platformio run' after creating a minimal .cpp file (copied below) it would not compile and complained about missing PubSubClient.h (error copied below, too)

I searched PlatformIO's libs and it has PubSubClient available as library 89 named 'PubSubClient'. Once I installed that, the example compiled with no errors.

I'm not sure if PlatformIO doesn't handle chained dependencies or if the AsyncMqttClient [346] isn't setup correctly, but I wasn't able to compile until I executed:
```
platformio lib install 89
```

Thanks for the great work! This does just about everything I need so far and a whole lot more!

### Compilation error
```
In file included from .pioenvs/d1_mini/Homie_ID555/Homie.hpp:3:0,
from .pioenvs/d1_mini/Homie_ID555/Homie.h:4,
from src/main.cpp:1:
.pioenvs/d1_mini/Homie_ID555/Homie/MqttClient.hpp:4:26: fatal error: PubSubClient.h: No such file or directory
#include <PubSubClient.h>
^
compilation terminated.
```

### main.cpp:
```
#include <Homie.h>

void setup() {
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

### Installation Output:
```
faraday:platformio-blink javamatte$ platformio lib install 555
Installing library [ 555 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #555 'Homie' has been successfully installed!
Installing dependencies:
Installing library [ 64 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #64 'Json' has been successfully installed!
Installing library [ 346 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #346 'AsyncMqttClient' has been successfully installed!
Installing dependencies:
Installing library [ 305 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #305 'ESPAsyncTCP' has been successfully installed!
Installing library [ 1106 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #1106 'Bounce2' has been successfully installed!

```

euphi, 2016-08-17:
It seeems that the wrong version of library.json is published at platformio!?

The one tagged with v1.5.0 is the correct one.

marvinroger, 2016-08-17:
There is something wrong with PlatformIO. [From the docs](http://docs.platformio.org/en/stable/librarymanager/config.html#version):

> version and repository fields are defined. The repository is hosted on GitHub or Bitbucket.

>PlatformIO Library Registry Crawler will lookup for release tag named as value of version or with v prefix (you do not need to pass this v prefix to the version field).

The latest version in the `library.json` is the 1.5.0, so it should use the `library.json` from the `v1.5.0` tag, which is the latest release. I think what's wrong is that PlatformIO uses the `library.json` from the latest git commit, not from the release the version field points to.

marvinroger, 2016-08-17:
This might not be a bug from PlatformIO, just something not clarified in the docs. Anyway, it should be fixed in a few days (waiting for the crawler).

commit 2f28f1dc (2016-08-17): Fix #129
closed 2016-08-17 by marvinroger