homieiot/homie-esp8266 #131: Homie Configurator: JSON values not trimmed

euphi, 2016-08-20:
The Homie configurator app does not trim the values to send. So it can happen that values contain trailing whitespaces.

Some values are validated (e.g. device-id), some not (e.g. WiFi password).

The app, or maybe even homie-esp8266 itself, should trim the values before sending (and before validating), because most Android keyboards adds a trailing space if a autocomplete feature is active.

marvinroger, 2016-08-20:
Alright, I never had such a problem with multiple keyboards (SwiftKey, Fleksy, native...), but trimming value might indeed avoid some trouble. :)

marvinroger, 2016-10-11:
Fixed in https://github.com/marvinroger/homie-esp8266-setup/commit/1db4599bcacf3ca7cfec334edeebae4a999926bb

closed 2016-10-11 by marvinroger