homieiot/homie-esp8266 #133: Sonoff: using reset button to switch relay

rokrodic, 2016-08-21:
Hi.
How to use Sonoff integrated button to manually switch the relay. The button is used for node reset in Homie.

`
#include <Homie.h>

const int PIN_RELAY = 12;
const int PIN_LED = 13;
const int PIN_BUTTON = 0;

HomieNode switchNode("switch", "switch");

bool switchOnHandler(String value) {
  if (value == "true") {
    digitalWrite(PIN_RELAY, HIGH);
    Homie.setNodeProperty(switchNode, "on", "true");
    Serial.println("Switch is on");
  } else if (value == "false") {
    digitalWrite(PIN_RELAY, LOW);
    Homie.setNodeProperty(switchNode, "on", "false");
    Serial.println("Switch is off");
  } else {
    return false;
  }
  return true;
}

void button() {
  
}
void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW);

  Homie_setFirmware("itead-sonoff", "1.0.0");
  Homie.setLedPin(PIN_LED, LOW);
  Homie.setResetTrigger(PIN_BUTTON, LOW, 30000);
  attachInterrupt(PIN_BUTTON, button, LOW);
  switchNode.subscribe("on", switchOnHandler);
  Homie.setup();
}

void loop() {
  Homie.loop();
}
`

Something like above code, but I do not think it is the best idea. Am I wrong? What is the solution in other Homie ESPs?

rokrodic, 2016-08-25:
Well here's the solution, but it would be nice to add a function to Homie something like:  Homie.setResetTrigger(PIN_BUTTON, LOW, 30000, mybuttonfunction, 1000);

void Setup() {
  .....
  Homie.setup();
    rstDebouncer.attach(PIN_BUTTON);
    rstDebouncer.interval(TOGGLE_KEYPRESS);
}

void loop() {
  if (millis() < LastToggle) LastToggle = 0;
  Homie.loop();
  boolean changed = rstDebouncer.update();
  if (changed) {
    if (rstDebouncer.read() == LOW) {
      if (millis() - LastToggle >= TOGGLE_RELAY) {
        ToggleCounter++;
        Serial.print(F("Toggle relay: "));
        Serial.println(ToggleCounter);
        int val = digitalRead(PIN_RELAY);
        if (val == HIGH) switchOnHandler("false"); else switchOnHandler("true");
        LastToggle = millis();
      }
    }
  }
}

marvinroger, 2016-08-27:
The v2.0.0 will provide a function to directly erase the configuration anyway, so you will be able to manage this kind of thing yourself. :smiley: 
Glad you solved it!

closed 2016-08-27 by marvinroger