homieiot/homie-esp8266 #134: Problem with i2c

mikecay, 2016-08-23:
Hi,

I am trying to use the wire.h library with homie 1.5.  The following example sketch works on a ESP-07.

`#include "Wire.h"
#include "Adafruit_INA219.h"

Adafruit_INA219 ina219;

void setup(void) 
{
  uint32_t currentFrequency;
    
  Serial.begin(115200);
  Serial.println("Hello!");
  
  Serial.println("Measuring voltage and current with INA219 ...");
  ina219.begin();
}

void loop(void) 
{
  float shuntvoltage = 0;
  float busvoltage = 0;
  float current_mA = 0;
  float loadvoltage = 0;

  shuntvoltage = ina219.getShuntVoltage_mV();
  busvoltage = ina219.getBusVoltage_V();
  current_mA = ina219.getCurrent_mA();
  loadvoltage = busvoltage + (shuntvoltage / 1000);
  
  Serial.print("Bus Voltage:   "); Serial.print(busvoltage); Serial.println(" V");
  Serial.print("Shunt Voltage: "); Serial.print(shuntvoltage); Serial.println(" mV");
  Serial.print("Load Voltage:  "); Serial.print(loadvoltage); Serial.println(" V");
  Serial.print("Current:       "); Serial.print(current_mA); Serial.println(" mA");
  Serial.println("");

  delay(2000);
}`

However, when I try to run pretty much the same sketch using the homie library, it does not work.  The homie sketch is as follows

`#include "Homie.h"
#include "Wire.h"
#include "Adafruit_INA219.h"

Adafruit_INA219 ina219;

HomieNode solarNode("solar", "voltage");

void loopHandler() {

  float shuntvoltage = 0;
  float busvoltage = 0;
  float current_mA = 0;
  float loadvoltage = 0;

  shuntvoltage = ina219.getShuntVoltage_mV();
  busvoltage = ina219.getBusVoltage_V();
  current_mA = ina219.getCurrent_mA();
  loadvoltage = busvoltage + (shuntvoltage / 1000);
  
  Serial.print("Bus Voltage:   "); Serial.print(busvoltage); Serial.println(" V");
  Serial.print("Shunt Voltage: "); Serial.print(shuntvoltage); Serial.println(" mV");
  Serial.print("Load Voltage:  "); Serial.print(loadvoltage); Serial.println(" V");
  Serial.print("Current:       "); Serial.print(current_mA); Serial.println(" mA");
  Serial.println("");

  delay(2000);
  
}

void setup(void) 
{
  uint32_t currentFrequency;
    
  Serial.begin(115200);
  Serial.println("Hello!");
  Serial.println("Measuring voltage and current with INA219 ...");

  ina219.begin();

  Homie.registerNode(solarNode);
  Homie.setLoopFunction(loopHandler);
  Homie.setup();

}

void loop(void) 
{
  Homie.loop();
}`

Any help would be appreciated

Mike

euphi, 2016-08-24:
Don't call delay() in the loophandler() or loop() function, if you run homie (or any other lib that needs frequent calls to their loop-Handler).

Just check, if enough time has passed at the beginning of loop and return oterwise.

E.g.
```
void loopHandler() {
   static uint32_t last = 0;
   Homie.loop();
   if (millis() - last < 2000) return;
   last = millis();
......

```

marvinroger, 2016-08-27:
Thanks @euphi for the help. @mikecay please report back if this solved the issue.

mikecay, 2016-08-28:
Hi,
 Yes, the suggestion worked. 

Thank you,

Mike

closed 2016-08-28 by mikecay