homieiot/homie-esp8266 #135: esp8266-01 problem

gubagu, 2016-08-24:
Marvin man fuckin' amzing project, and the heavy lifting you've done WOW !!!

I run the Homi server it worked just out of the box. Connected it to my MQTT broker - everything worked out of the box.
Then I installed the example code on on of my esp8266-01 and nothing happened. It did not create AP as it is said in the getting started manual and the serial had also nothing to say. Once i got some error stack output but that was once.

What could it be? 

Tell me if you need additional info on the issue.

Thanks in advance man.

Cheerz, 
Guby

benzino77, 2016-08-24:
https://homie-esp8266.readme.io/docs/built-in-led
Try to disable led feedback

gubagu, 2016-08-24:
@benzino77  will try and let you know. Chgeerz mate

euphi, 2016-08-24:
Activate Logging to Serial in the setup() function.
```

Serial.begin(115200);
homie.setLogging(true);
```

See, if there are any error messages when booting ConfigurationMode.

gubagu, 2016-08-24:
Hey @benzino77 here is the output when trying to compile :

````
Arduino: 1.6.10 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200, 4M (1M SPIFFS)"

C:\Users\Zarro\AppData\Local\Temp\arduino_modified_sketch_360\LightOnOff.ino: In function 'void setup()':

LightOnOff:24: error: 'class HomieInternals::HomieClass' has no member named 'disableLedFeedback'

   Homie.disableLedFeedback(); // before Homie.setup()
         ^

exit status 1
'class HomieInternals::HomieClass' has no member named 'disableLedFeedback'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

marvinroger, 2016-08-24:
The documentation linked above is for the upcoming v2. The current
documentation is there:
https://github.com/marvinroger/homie-esp8266/blob/v1.5.0/docs/9.-Troubleshooting.md
:)

gubagu, 2016-08-24:
@marvinroger this got me little bit further...
[image]

marvinroger, 2016-08-27:
* Format your SPIFFS:

```c++
#include "FS.h"

void setup() {
  Serial.begin(115200);
  Serial.println("");
  delay(1000);
  Serial.println("Mounting FS...");

  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount file system");
    return;
  }

  if (!SPIFFS.format()) {
    Serial.println("Failed to format");
  } else {
    Serial.println("File system formatted");
  }
}

void loop() {
}
```

* Flash the following Homie sketch:

```c++
#include <Homie.h>

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Homie.enableBuiltInLedIndicator(false);
  Homie.disableResetTrigger();
  Homie.setFirmware("test", "1.0.0");
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

Does your serial monitor print something then?

gubagu, 2016-08-29:
Hey there,
The first sketch has the same result as my previous comment.

clough42, 2016-08-29:
@gubagu It looks like you're compiling for an ESP12 module with a 4MB (32Mb) flash chip.  Most ESP-01 modules come with either 512KB (4Mb) or 1MB (8Mb) flash. 

I had a similar problem until I switched board definitions to a 512KB ESP01.

gubagu, 2016-08-30:
Hey @clough42  indee that was it ! Thanks a bunch !

closed 2016-08-30 by gubagu