homieiot/homie-esp8266 #136: Update config via mqtt

benzino77, 2016-08-25:
$implementation/version: 2.0.0

I think MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE in Limits.h is calculated wrong.
When I'am trying to update config via publishing new config to $implementation/config/set I get error:

> Invalid or too big JSON

Here is the config which I publish to $implementation/config/set:
```
{
  "name": "Well sensor",
  "wifi": {
    "ssid": "myssid",
    "password": "sueprsec"
  },
  "mqtt": {
    "host": "192.168.0.101",
    "port": 1883,
    "auth": true,
    "username": "user",
    "password": "supersec"	
  },
  "ota": {
    "enabled": true
  },
  "settings": {
    "test0": 1800,
    "interval": 10,
    "test1": "value1"
  }
}
```
Changing line:
`const uint16_t MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE = JSON_OBJECT_SIZE(6) + JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(6) + JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(10);  // Max 5 elements at root, 2 elements in nested, etc... the last 10 means 10 custom settings max`

to:

`const uint16_t MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE = JSON_OBJECT_SIZE(6) + JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(6) + JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(10`**+14**`);  // Max 5 elements at root, 2 elements in nested, etc... the last 10 means 10 custom settings max`

solves the problem.

benzino77, 2016-08-25:
I just find (!!) that I can update config incrementally. This way I can update only `settings `section.

vladbabii, 2016-08-26:
Didn't know I could update the configuration this way. Thank you!

marvinroger, 2016-08-27:
Weird, the above JSON should fit within the buffer... I'll check!

marvinroger, 2016-10-17:
This should be fixed now.

closed 2016-10-17 by marvinroger