homieiot/homie-esp8266 #140: WPS connections for Homie?

jpmens, 2016-08-30:
If I understand WPS, it would allow Homie devices to autoconfigure themselves for the WiFi connection, and this would be something grand to have in Homie: we could install a device, push a button on the WiFi AP and have Homie auto-connect to Wifi (and subsequently, via mdns to MQTT).

In #87 there is mention of WPS for initiating WiFi connection to the AP, and judging by [this code](https://gist.github.com/copa2/fcc718c6549721c210d614a325271389), Arduino ESP8266 is actually capable of doing WPS.

Has anybody recently tried implementing this for Homie-esp8266?

marvinroger, 2016-08-31:
The problem is the current WPS implementation is synchronous. I've filled an issue on https://github.com/esp8266/Arduino/issues/2464, let's see if this is doable.

But even if doable, how would the device be configured? Via mdns, we would have the IP/port, but what about the device id, base topic, auth... ?

jpmens, 2016-08-31:
Bah, those pesky questions ... I was hoping you'd just give answers! 😆 

Yes, we'd have IP/port for the broker. Device ID defaults to the hex serial which is fine, and authentication (in this particular case) is not really needed. And we already have a default base topic (`devices/<deviceID>`) in the _convention_, don't we?

shogsbro, 2016-08-31:
Hmm, mDNS network discovery? Could use SRV records for MQTT broker IP/port, but wouldn't help with authentication credentials.

stufisher, 2016-10-15:
I have been thinking about this, this is really the stumbling block to making this "general public" friendly. Could we define some sort of wps like protocol for mqtt? Even if it required you to push a physical button on the broker (say a pi in my case), that would be better than entering credentials on a config screen.

timpur, 2018-04-25:
@nicola-lunghi how do you `mqtt.mylocalpi` onto the device initially? also username and password?