homieiot/homie-esp8266 #142: Feature Proposal: Listen to other topics -> Broadcast channel?

euphi, 2016-08-31:
As already proposed in #80 (but closed due to non-feasibility with the old PubSubClient), there should be an interface to subscribe to other devices/nodes.

My use case is to have a broadcast channel to send alarms and messages to all other devices.

In detail: I have a central OpenHAB server running with a local MQTT. The OpenHAB server should be able to publish MQTT message to all connected (Homie-)device, e.g. to trigger an alarm. I have some nodes with buzzer and display so they could show even some alarm messages.

You may even define a Broadcast channel in the Homie convention, e.g. `$broadcast/<level>` and `$alarm/<type>` ?
So, in my idea, any HomieNode shall be able to subscribe to broadcasts and alarms and gets called back.
Personally I would prefer an extra input-handlers e.g. `virtual void handleBroadcast(EBCstLevel level, String text)`.

If I get some feedback here, I will develop a proposed solution in my local fork.

marvinroger, 2016-09-06:
What about a way for you to subscribe to any topics you want? This way you can do pretty much everything. The current AsyncMqttClient implementation only allow one callback for messages, so you cannot override it, as Homie already handles messages. But I plan (https://github.com/marvinroger/async-mqtt-client/issues/21) to allow multiple callbacks. Would it be alright for you?

The broadcast channel is not a bad idea, but people will probably want more and more MQTT related-stuff anyway, so a complete access to the underlying MQTT client would solve all these problems.

euphi, 2016-09-06:
I think, broadcast channels should be part of Homie (the ESP implementation and the convention), but subscription to arbitrary topics should not. It's always good to have a quite strict API to force people to follow the convention.

So to allow arbitrary subscription, Homie should only offer a pointer (or reference) to the AsyncMqttClient instance, so anybody can bypass Homie and implement there own callback (as soon as 
https://github.com/marvinroger/async-mqtt-client/issues/21 is implemented).

I will be on a business trip and then on holidays during the next weeks. When I return I will develop my local fork to add support for a broadcast channel.

marvinroger, 2016-09-07:
The abitrary topics part would not be part of the convention, and yes to achieve arbitrary subs, you can use `Homie.getMqttClient()` which returns a reference to `AsyncMqttClient`.

Let's implement a broadcast channel, then!

marvinroger, 2016-09-07:
Have great holidays, by the way. :wink:

marvinroger, 2016-09-07:
[Broadcast channel in the Homie convention](https://github.com/marvinroger/homie/tree/cd84f93ad8a3d9da838d7e6f0402241a54b376e4#broadcast-channel).

Nzbuu, 2016-09-07:
Why `devices/$broadcast/#`? Would `devices/all/#` or `devices/broadcast/#` be more natural?

marvinroger, 2016-09-07:
Everything "special" in Homie starts with a `$`. Think parsing. When you parse the topic, you only have to check if the subtopic starts with a `$` to know it is a special one.

euphi, 2016-09-20:
Why `devices/$broadcast/` and not just `$broadcast/` ? A broadcast is not a device, so $broadcast should not be in a subtree of `devices`. If you do not want to place it at toplevel, maybe you could introduce a new top-level hierarchie `$homie/` (or `homie/`).

marvinroger, 2016-09-20:
Well, actually `devices/` is the base topic and is configurable, so you can use pretty much any root topics you want. I'll indeed maybe change the default topic from `devices/` to `homie/`.

euphi, 2016-09-27:
I started to implement, now I wonder how the `_mqttPayloadBuffer` is ever deleted in BootNormal? See my comment to commit c46704a611896ef1586defc2df71a22f3696e1e5.

closed 2016-10-11 by marvinroger

shogsbro, 2016-11-01:
Marvin, could I suggest that the readme.io documentation is updated to clarify where in the MQTT topic namespace broadcasts will be published? 

I would prefer anchoring it from the configured base root. Imagine your MQTT server has several generations of firmware running; hardcoding `homie/` as the broadcast path would force you into running several MQTT brokers, with some judicious replication of selected topics.

commit 85769fd0 (2016-10-11): :sparkles: Add broadcast channel - fix #142 (#161)

* First proof of concept for $broadcast feature

- $broadcast is handled and globalInputHandler is called
- still many TODOs, e.g. support for other callbacks or splitted MQTT
message (make use of payload buffer)

* $broadcast - use payload buffer

+ minor improvements

* Added $broadcast handler

+ added comments to BootNormal.cpp to explain handling of incoming MQTT
message