homieiot/homie-esp8266 #143: Base64-encode MQTT OTA payloads?

jpmens, 2016-09-02:
Homie 2.0 will support performing OTA updates using a raw firmware binary published retained to an MQTT topic.

Should we consider base64-encoding the firmware blob at that retained topic? Yes, it would increase the message size (I don't really think that'd be an issue), but it would have the advantage that normal user utilities subscribed to the same topic (I'm thinking `mosquitto_sub`, et. al.) wouldn't be "bothered" by the binary message. (Think `mosquitto_sub` printing to _stdout_ and messing up terminals. :-)

marvinroger, 2016-09-02:
The increase in size is like 33% so that's not much of an issue. But still, more data to receive means more risk to receive a bad firmware, which is a critical piece of software. I mean, if the only benefit is not to mess up the terminal, I am not sure this is a good idea to add base64 decoding/network bandwith overhead... Is that really important to you?

jpmens, 2016-09-03:
Not very, no.

closed 2016-09-03 by jpmens

MaxChinni, 2016-09-05:
You can prevent mess on the terminal with something like

    mosquitto_sub -t '#' -T 'homie/devices/+/$ota/payload'

Here's the related `man` page parts

       -t, --topic
           The MQTT topic to subscribe to. See mqtt(7) for more information on MQTT topics.

           This option may be repeated to subscribe to multiple topics.

       -T, --filter-out
           Suppress printing of topics that match the filter. This allows subscribing to a wildcard topic and only printing a
           partial set of the wildcard hierarchy.