homieiot/homie-esp8266 #146: Deep sleep function without interfering with the homie module functions

gubagu, 2016-09-08:
Question:

Is it possible to use esp8266's "ESP.deepSleep(microseconds, mode)" function without interfering with the homie module functions (https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#esp-specific-apis) ?

I have in mind enegy efficient sensor that lets say get measurements publish them to the broker and goes in deepsleep again.

Furhter more i noticed that home modules report their availability - alive and probably to turn off that function in the mode with deep sleep.

Thanks in advance !

Cheerz, Guby

marvinroger, 2016-09-09:
Yes, you can use it. The upcoming version will allow you to be notified when one of your message is received by the broker. So you will only have to deep sleep when you receive the notification.

The initial sending of information is only a matter of milliseconds, whereas you need to wait a second to connect to the Wi-Fi / broker anyway, so there's no need to sacrifice this feature to gain like 30ms out of a second or two.

closed 2017-11-03 by gubagu

luebbe, 2017-11-03:
This may also be of interest to you: [Homie and light sleep - How it is done](https://github.com/marvinroger/homie-esp8266/issues/412).
I needed a sensor that runs on batteries, consumes as little energy as possible but can be woken up by an interrupt or button press without having reboot completely.