homieiot/homie-esp8266 #147: OTA Configuration Updates

onezero1010101, 2016-09-09:
I have OTA firmware updates working, and am now trying to update the config.json file on the device thru the mqtt you have listed.  I must not understand the process you have listed in the readme, I am trying to read the config by using the $implementation/config but never get anything back.  Is there something specific I publish to that mqtt to get it to send the config?  From what I understand this can be done in normal mode, but I get no response.  I also tried to push json to change device name by itself to $implementation/config/set but didn't get any change or response there either.  My base_topic is set to sensor and the device ID is dev1, so I'm using sensor/dev1 for $implementation which I assume is correct.  Hopefully I'm just confused on the process.  Thanks, I love the project by the way.

marvinroger, 2016-09-09:
Which Homie for ESP8266 version do you use?

onezero1010101, 2016-09-09:
I am using 1.5 from the public release.  From further digging thru the issues logs on here, I think I found my problem.  I have been using the homie-esp8266.readme.io documentation because everything was broken I tried to use on github and said it was now on there.  But that is for 2.0.0 isn't it.  So what I am trying to do isn't possible yet.  

Sorry for the confusion.  I am new to github still, there's no where to beta test the new 2.0 that the public has access to is there?  Not a big deal, just curious.

marvinroger, 2016-09-09:
So yes, the docs for v1.5.0 are [here](https://github.com/marvinroger/homie-esp8266/blob/v1.5.0/docs/index.md)

You can beta-test the 2.0.0 (which is actually not even released as beta) by cloning the repository. The latest git available is the 2.0.0.

closed 2016-10-07 by marvinroger