homieiot/homie-esp8266 #148: Examples won't compile (version number needs to be bumped?)

zouden, 2016-09-10:
Hi, I'm very excited to start building Homie devices using my ESP8266 boards! I've just installed Homie using PlatformIO's library manager, which works, but the LED strip controller example won't compile.

    error: 'HomieRange' was not declared in this scope

My local copy of Homie 1.5.0 doesn't have Homierange.cpp. I think this is because when you added HomieRange, and updated the examples to use it, you didn't change the version number. As a result the version in PlatformIO's repository is out of date.

zouden, 2016-09-10:
Pinging @ivankravets as I just noticed you're a contributor here too.

ivankravets, 2016-09-10:
You use stable 1.5.0 version where this HomieRange has not been implemented yet. If you want to try development version of this library, you should use GitHub URL or path to archive with latest sources. For example,
```
platformio lib install https://github.com/marvinroger/homie-esp8266.git
platformio lib install https://github.com/marvinroger/homie-esp8266/archive/master.zip
```

In any case, you should deinstall previous stable version via `platformio lib uninstall Homie`.
I recommend to use first option with GIT VCS. In this case, `platformio lib update Homie` will automatically sync library with thisrepository.

zouden, 2016-09-10:
Great, thanks a lot! It compiles now using the latest source from the repository.

marvinroger, 2016-09-10:
Thanks @ivankravets for the explanation
@zouden the code in the master branch is the 2.0.0 one, which is not yet released. As a result, PlatformIO does not know `HomieRange`. It's not a mistake, it's just the v2 [is not ready](https://github.com/marvinroger/homie-esp8266/milestone/6).

Examples are bundled within each release. For PlatformIO, I don't know if there's a built-in way to show examples from the CLI, [but the examples on the Homie PlatformIO page](http://platformio.org/lib/show/555/Homie) are the right ones, the 1.5.0 examples.

ivankravets, 2016-09-10:
Examples are located in the same directory where library is installed. @zouden check it via `pio lib list`, where you will Library Storage path, follow to `Homie` and check `examples` folder.

closed 2016-10-07 by marvinroger