homieiot/homie-esp8266 #152: Node doesn't receive MQTT messages

zouden, 2016-09-13:
I'm using v2.0.0 on Wemos D1, just trying to get Homie working with the readme code example (lightNode). [Here's the code ](http://pastebin.com/AeyD5AU9)on pastebin.

The device boots, config.json is read, and my MQTT broker gets the status topics like **devices/strip2/$uptime** and **devices/strip2/light/$properties** etc. But if I publish a message to the actual property, **devices/strip2/light/on**, it isn't received by the node: there's nothing printed on the serial port. Only the regular "Sending uptime..." messages. 

Any ideas? Is 2.0.0 unusable? I get no compiler warnings.

zouden, 2016-09-13:
Problem solved: the actual topic for changing the value should be /devices/strip2/light/on/set. I see that's in the Homie readme rather than in Homie-esp8266.

closed 2016-09-13 by zouden

marvinroger, 2016-09-14:
That's why it's marked as a requirement to understand the convention. :)