homieiot/homie-esp8266 #153: [sharing] homie-control

stufisher, 2016-09-16:
This isn't an issue as such, just a pointer to a web application i've put together for the homie framework: https://github.com/stufisher/homie-control (some screenshots here: https://github.com/stufisher/homie-control/tree/master/client/screenshots)

Gives you scheduling, profiles, triggers, and history for homie devices. Work ongoing, pretty developer orientated at the moment...

benzino77, 2016-09-16:
WOW! Screenshots looks pretty nice.
Defenetly I'll check it!

marvinroger, 2016-09-18:
Awesome, thanks for that! I added your project to the [Community projects](https://homie-esp8266.readme.io/docs/community-projects) page of the docs. Feel free to suggest edits to this page if you want to. 😉

gubagu, 2016-11-04:
@stufisher Looks great man !

stritti, 2019-03-23:
I close this issue because the project is already referred within the docs: https://homieiot.github.io/homie-esp8266/docs/stable/others/community-projects/

closed 2019-03-23 by stritti