homieiot/homie-esp8266 #154: A way to store values between resets?

alphalove, 2016-09-17:
Hello,
I've tried a few other MQTT implementations for the 8266, but this one looks great, well done!

Is it possible to add functionality within this framework to store values that will persist between resets?

My purpose is controlling lights and power sockets and I'd like the devices to set the stored / previous state should they be reset (and can't reach the MQTT broker etc.).

Perhaps even a put() similar to the get() function for the config values?

Thanks again!

(*sorry, I wasn't sure how to tag this as a 'feature request' rather than an issue!)

jpmens, 2016-09-17:
One way of doing this is to use the SPIFFS (sp?), i.e. opening, writing, reading a file on the file system of the ESP8266. I suppose this could be done from within the framework, but it might be difficult to "coordinate" the locally stored values with values obtained from retained topics on a connected MQTT broker. For example, device comes online, reads "lamp ON" from SPIFFS, and then possibly gets "lamp OFF" from the broker; this would lead to flickering.

alphalove, 2016-09-17:
Hello jpmens,
Thanks for the quick reply!  I'm new to the use of SPIFFS, but I'll have a look now.  Should it be used similarly to how you might use EEPROM.write() in aduinos?  Or would it be considered over kill?

The status of the switch should be written to permanent memory every time it changes.  I suppose  a persistent message from the broker could be conflicting and cause flickering... But I think I'd use the setSetupFunction to send the stored value to the broker on device startup, therefore syncing them back up if they were different.

Thanks again.

benzino77, 2016-09-17:
You could try:
`ESP.rtcUserMemoryWrite`
`ESP.rtcUserMemoryRead`

marvinroger, 2016-09-18:
As said above, an Homie sketch is also a normal Arduino sketch, so you can use EEPROM, SPIFFS, RTC memory etc as you would normally do. If you want a simple approach, I would recommend you [Embedis](https://github.com/thingSoC/embedis), which allows you to do:

```cpp
String my_value;
Embedis::get("key", my_value);
```

marvinroger, 2016-10-07:
Closing this issue as I am trying to tidy up the issue tracker for the v2. :)

closed 2016-10-07 by marvinroger