homieiot/homie-esp8266 #155: BME280 + Homie not reading values.

billm950, 2016-09-17:
I been trying to get my BME280 Temperature Sensor connected with Homie.  Everything works great.  Except when I apply the working code that is reading the BME280 values, the values all come back null. 

If I remove the homie references, values print nicely on the serial.

What am I doing wrong, replacing with DHT22 library, works perfectly fine.

`
#include <Wire.h>     
#include <BME280.h>
#include <Homie.h>

#define FW_NAME "temp"
#define FW_VERSION "1.0.0"

BME280 bme;
bool metric = false;

HomieNode temperatureNode("temperature", "temperature");
HomieNode humidityNode("humidity", "humidity");
HomieNode pressureNode("pressure", "pressure");
HomieNode altitudeNode("altitude", "altitude");
HomieNode dewpointNode("dewpoint", "dewpoint");

const int SEND_INTERVAL = 60;
unsigned long lastDataSent = 0;

void loopHandler() {
  if (millis() - lastDataSent >= SEND_INTERVAL * 1000UL || lastDataSent == 0) {
    float temp(NAN), hum(NAN), pres(NAN);
    uint8_t pressureUnit(2);   // unit: B000 = Pa, B001 = hPa, B010 = Hg, B011 = atm, B100 = bar, B101 = torr, B110 = N/m^2, B111 = psi
    bme.ReadData(pres, temp, hum, metric, pressureUnit);                // Parameters: (float& pressure, float& temp, float& humidity, bool hPa = true, bool celsius = false)

    Serial.print("Temp: ");
    Serial.print(temp);
    Serial.print("°"+ String(metric ? 'C' :'F'));
    Serial.print("\t\tHumidity: ");
    Serial.print(hum);
    Serial.print("% RH");
    Serial.print("\t\tPressure: ");
    Serial.print(pres);
    Serial.print(" atm");

    float altitude = bme.CalculateAltitude(metric);
    float dewPoint = bme.CalculateDewPoint(metric);
    Serial.print("\t\tAltitude: ");
    Serial.print(altitude);
    Serial.print((metric ? "m" : "ft"));
    Serial.print("\t\tDew point: ");
    Serial.print(dewPoint);
    Serial.println("°"+ String(metric ? 'C' :'F'));

    Homie.setNodeProperty(temperatureNode, "degrees").send(String(temp));
    Homie.setNodeProperty(humidityNode, "relative").send(String(hum));
    Homie.setNodeProperty(pressureNode, "atm").send(String(pres));
    Homie.setNodeProperty(altitudeNode, "ft").send(String(altitude));
    Homie.setNodeProperty(dewpointNode, "degrees").send(String(dewPoint));

    lastDataSent = millis();
  }
}

void setupHandler() {

  Homie.setNodeProperty(temperatureNode, "unit").setRetained(true).send("f");  
  Homie.setNodeProperty(humidityNode, "unit").setRetained(true).send("%");  
  Homie.setNodeProperty(pressureNode, "unit").setRetained(true).send("atm");  
  Homie.setNodeProperty(altitudeNode, "unit").setRetained(true).send("ft");  
  Homie.setNodeProperty(dewpointNode, "unit").setRetained(true).send("f");
  
  temperatureNode.advertise("degrees");
  temperatureNode.advertise("unit");
  humidityNode.advertise("relative");
  humidityNode.advertise("unit");
  pressureNode.advertise("atm");
  pressureNode.advertise("unit");
  altitudeNode.advertise("ft");
  altitudeNode.advertise("unit");
  dewpointNode.advertise("degrees");
  dewpointNode.advertise("unit");

}

void setup() {
  Serial.begin(115200);
  Wire.begin(0,2);         
  while(!Serial) {} // Wait
  while(!bme.begin()){
    Serial.println("Could not find BME280 sensor!");
    delay(1000);
  }
  
  Homie_setFirmware(FW_NAME, FW_VERSION);
  Homie.setSetupFunction(setupHandler).setLoopFunction(loopHandler);
    
  Homie.setup();
}

void loop() {
  Homie.loop();
}
`

marvinroger, 2016-09-18:
I honestly have no idea of what's going on. The `loopHandler()` is called from the `Homie.loop()` function, itself called from `loop()`. So if it works in `loop()`, it should work in `loopHandler()`.

billm950, 2016-09-18:
if I run the sensor start up code inside    `case HomieEvent::NORMAL_MODE:`, BME280 is detected fine, but no readings.  If I run inside after `WIFI_CONNECTED` or `MQTT_CONNECTED`, BME280 fails to be detected

pdominique, 2016-09-19:
This code works for me (Homie 1.5 I believe), maybe you can try it to see if this works for you too:

```
// BME280   Wemos D1 Mini
// VIN -->  3V3
// GND -->  G
// SCL -->  D1
// SDA -->  D2

#include <Homie.h>
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

#define SEALEVELPRESSURE_HPA (1013.25)

Adafruit_BME280 bme; // I2C

const int TEMPERATURE_INTERVAL = 5;
const int HUMIDITY_INTERVAL = 10;
const int PRESSURE_INTERVAL = 15;

unsigned long lastTemperatureSent = 0;
unsigned long lastHumiditySent = 0;
unsigned long lastPressureSent = 0;

HomieNode weatherNode("weather", "weather");

void setupHandler() {
  if (!bme.begin()) {
    Serial.println("Could not find a valid BME280 sensor, check wiring!");
    while (1);
  }
}

void loopHandler() {
  // Temperature
  if (millis() - lastTemperatureSent >= TEMPERATURE_INTERVAL * 1000UL || lastTemperatureSent == 0) {
    float temperature = bme.readTemperature();
    Serial.print("Temperature = ");
    Serial.print(temperature);
    Serial.println(" *C");
    if (Homie.setNodeProperty(weatherNode, "temperature", String(temperature), true)) {
      lastTemperatureSent = millis();
    } else {
      Serial.println("Sending failed");
    }
  }
  // Humidity
  if (millis() - lastHumiditySent >= HUMIDITY_INTERVAL * 1000UL || lastHumiditySent == 0) {
    float humidity = bme.readHumidity();
    Serial.print("Humidity = ");
    Serial.print(humidity);
    Serial.println(" %");
    if (Homie.setNodeProperty(weatherNode, "humidity", String(humidity), true)) {
      lastHumiditySent = millis();
    } else {
      Serial.println("Sending failed");
    }
  }
  // Pressure
  if (millis() - lastPressureSent >= PRESSURE_INTERVAL * 1000UL || lastPressureSent == 0) {
    float pressure = bme.readPressure() / 100.0F;
    Serial.print("Pressure = ");
    Serial.print(pressure);
    Serial.println(" hPa");
    if (Homie.setNodeProperty(weatherNode, "pressure", String(pressure), true)) {
      lastPressureSent = millis();
    } else {
      Serial.println("Sending failed");
    }
  }
}

void setup() {
  Homie.setFirmware("awesome-weather", "1.0.0");
  Homie.setSetupFunction(setupHandler);
  Homie.setLoopFunction(loopHandler);
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

koen01, 2017-12-28:
@billm950
did you ever get this working? I'm having similar issues but then with a BH1750 sensor via I2C. 
Not using homie-esp this sensor works as intended. 
Using homie-esp the sensor returns error values...

@marvinroger 
I can post code if you want?
i tried several BH1750 libraries.

stritti, 2019-04-02:
Probably test more current version of Homie-ESP8266. If it is still an issue, please reopen.

closed 2019-04-02 by stritti

nemidiy, 2019-11-17:
FWIW, I was having the same issue. For some reason if I try to interact with my i2c sensor 
straight from the loop() function it does not work, instead, if I define a loop handler function 
and pass it on to Homie and interact from there, it works.

The sensor I'm using is not BME280 but this https://www.atlas-scientific.com/product_pages/circuits/ezo_flow.html so this seems to be about the interaction between homie and the Wire library and not a specific sensor.

This sensor has a led that blinks when it gets a command so it's easy to know what's going on.
As mentioned before, if I do all the Wire stuff on the loop() function after the Homie.loop() call instead,
the led in the sensor does not even blink and readings are null, this suggests that the command somehow is not being sent over the i2c bus. I have no clue why ...

Here is the code that worked form me :

```c++
#include <Homie.h>
#include <Wire.h>

HomieNode sensorNode("sensor", "Sensor", "sensor");

void loopHandler() {
    Wire.beginTransmission(104);
    Wire.write("R\0");
    Wire.endTransmission();
    delay(500);
    Wire.requestFrom(104, 20, 1);
    //the first byte is the response code, we read this separately.
    byte code = Wire.read();
    char buffer[20];
    if(code == 1){
      Serial.println("Success");
      byte i = 0;
      while (Wire.available()) {
        buffer[i] = Wire.read();
        if (buffer[i] == 0){
          Wire.endTransmission();
          break;
        }
        ++i;
      }
      Serial.print("Response : ");
      Serial.println(buffer);
      sensorNode.setProperty("value").send(String(buffer));
    }else if (code == 2){
      Serial.println("Failure");
    }else if (code == 254){
      Serial.println("Not finished");
    }else if (code == 255){
      Serial.println("No data");
    }
}

void setup() {

  Serial.begin(115200);
  Serial << endl << endl;

  Wire.begin();

  Homie_setFirmware("i2c-test", "1.0.0");
  Homie.setLoopFunction(loopHandler);

  sensorNode.advertise("value").setName("Value").setDatatype("double");

  Homie.setup();
}

void loop(){
    Homie.loop();
    // without this delay it will not connect to the MQTT broker :S
    delay(100);
}
```
Here is my platformio setting:
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags =
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
upload_speed = 115200
lib_ldf_mode = deep
lib_deps =
    ArduinoJson@6.13.0
    https://github.com/homieiot/homie-esp8266.git#develop-v3
```

kleini, 2019-11-18:
@nemidiy How did you start to develop the code? If you already have problems connecting to MQTT broker, then you should start with the most simple code, just the Homie stuff for itself and not with a large block of code, that can have a lot of issues. Get this little thing, just the Homie stuff working reliably. If this is solved and works good, you can start to integrate reading some sensor, but not the other way round. I have the impression, that people just copy & paste stuff and beg to god, that this might work.

amayii0, 2019-11-18:
Be careful when using delay(), the watch dog doesn't like it much AFAIKR
In your loop, it might be easier to split between actions : reading and interpreting (if code == stuff)
Not sure about the Wire.beginTransmission in the loop, looks like it'd be better in a setup function

Also, it's way easier to use logging and read the console.
Here is an example from my DHT22 test repo : https://github.com/amayii0/Homie-DHT22/blob/master/src/main.cpp#L31

luebbe, 2019-11-18:
don't use delay() in the loop. If you want to take a measurement every second use something like:

```
void loopHandler() {
  if (millis() - lastMeasureSent >= MEASURE_INTERVAL * 1000UL || lastMeasureSent == 0) {
... your code goes here ...
    lastMeasureSent = millis();
  }
}

```

nemidiy, 2019-11-18:
hey thanks for the answers.

@kleini you are right, I should not have skipped steps. Anyways since I started using homie some time ago I have been having issues connecting to the MQTT broker (mosquitto), the only thing that solved it was that nasty delay call on the main loop. Oddly enough the delay makes it extremely stable, like in days without disconnecting. I will create a separate ticket for that with a bare bones sketch to illustrate what's going on and maybe you can help out. PS: I am an atheist so I don't beg to god ;)

@amayii0, about the delay you are right, about the Wire.beginTransmission that would not be possible since you need to do that every time you send a command over the I2C bus marking the beging and the end of the payload beign sent.

@luebbe In fact in my real implementation I am doing exactly that, different version but I check for the right time to query instead of using the delay. This was just an ugly sketch to try and show the issue, my bad there.