homieiot/homie-esp8266 #156: PrepareForSleep in v2.0

mikecay, 2016-09-18:
Hi,

I installed version 2.0 in hopes to use the PrepareForSleep on a battery powered node.  I am using the sketch below.  I receive the $online false message but the node never wakes up.  Another thing also, I never get anything on my serial monitor.  I have this running an an ESP-7.  Any help would be appreciated.  

`#include <Homie.h>
#include <Wire.h>
#include <Adafruit_INA219.h>

#define FW_NAME "vtest"
#define FW_VERSION "1.0.1"

/* Magic sequence for Autodetectable Binary Upload */
const char *__FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" FW_NAME "\x93\x44\x6b\xa7\x75";
const char *__FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" FW_VERSION "\xb0\x30\x48\xd4\x1a";
/* End of magic sequence for Autodetectable Binary Upload */

unsigned long currentTime = 120 * 1000UL;   
unsigned long readTime = 0;
unsigned long readDelay = currentTime;
float busvoltage = 0;

Adafruit_INA219 ina219_A;

HomieNode solarNode("solar", "voltage");

void setupHandler() {
  Serial.println("Node is awake");
  Homie.setNodeProperty(solarNode, "state").send("awake");

}

void onHomieEvent(HomieEvent event) {
  switch(event) {
    case HomieEvent::READY_FOR_SLEEP:
      Serial.println("Going to sleep");
      ESP.deepSleep(15 * 1000000);
      break;
  }
}
void loopHandler() {
  if (currentTime - readTime >= readdelay) {
    Homie.setNodeProperty(solarNode, "state").send("in loop");
    busvoltage = ina219_A.getBusVoltage_V();
    Homie.setNodeProperty(solarNode, "vLoad").send(String(busvoltage));
    if (busvoltage < 6) {
      Serial.println("Preparing for sleep");
      Homie.setNodeProperty(solarNode, "state").send("Preparing for sleep");
      Homie.prepareForSleep();
    }
    readTime = millis();
  }
  currentTime = millis();
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  Wire.begin();
  ina219_A.begin();  // Initialize board (default address 0x40)
 
  Homie_setFirmware(FW_NAME, FW_VERSION);
  Homie.setSetupFunction(setupHandler);
  Homie.setLoopFunction(loopHandler);
  Homie.onEvent(onHomieEvent);
  Homie.setup();
}

void loop() {
  Homie.loop();
}

benzino77, 2016-09-18:
Do you have GPIO16 and RST wired?

mikecay, 2016-09-20:
Thank you, that was the problem. I remember reading about that but completely forgot to do it.

Mike

closed 2016-09-20 by mikecay