homieiot/homie-esp8266 #157: Homie breaks Serial setup

euphi, 2016-09-24:
For me, Homie makes the Serial configuration invalid (changed baudrate?). So it is necessary to call (again) `Serial.begin()` after `Homie.setup()`.

It is also not possible to see Homie debug output during setup (there should be at least 'Triggering NORMAL_MODE event...'). After setup, also the homie messages are fine.

The following setup
```
void setup() {
	Serial.begin(115200);
	Serial.println("Starting Heizungscontroller");
	Serial.flush();
	Homie.disableResetTrigger();
	Homie.setup();
	Serial.begin(115200);
	Serial.println("Finished Setup");
	Serial.flush();
}
```

results in the following output (including the noise from the bootloader at different baudrate):

```
rl␀l��|␀�l�|␂␌␌␌�␌l�␌b|��␂�␒�r�b�␌b��nn�lnn���␌b␜p��lrlrlp�n�␐␂␌␌�␌l␌��␌␌␌b␌n�|␂l�␌␌�b��nn�␀l��l`␂�␒␒nn␌l`␂␎␂nr���n␌␌b␌�␎l␎r��n␌␌b␌�␎l�r�����␌��l`␂��n�␂�*W�ѥ���
Heizungscontroller
T�␙���͡���Setup
✔ Wi-Fi connected
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_CONNECTED event...
3166: DEBUG/RelaisNode:Ready
3166: DEBUG/RelaisNode::updateRelais():Value: 0
sendI2C: Write value ffff.
Calling setup function...
Sending Wi-Fi signal quality (46%)...
Sending uptime (3s)...
```

You see, the Serial output of Homie seems to be written at a different baudrate. I investigated a little bit in the homie sourcecode, but up to now I have no clue what is happening?

I'm using latest git version and experienced the same behavior several weeks ago (but did not invastigated it).

marvinroger, 2016-09-26:
`Serial.begin()` is not called at all inside the library, so I have no idea what's going on here. I don't experience this, though. This is probably not related to Homie directly. Can you try to disable I2C in your sketch, to see if it has any negative effect?

euphi, 2016-09-26:
I2C is not initialized in global setup() nor in Homie::setup() (nor any method called by it).

Wire::begin() is already called *before* during the initialization phase, when  the related HomieNode is constructed. (My [RelaisNode](https://github.com/euphi/HomieNodeCollection/blob/master/src/RelaisNode.h) calls in its constructor the one of a  [PCF8575 class](https://github.com/euphi/HomieNodeCollection/blob/master/src/PCF8575.cpp).

marvinroger, 2016-10-07:
Does it happen everytime? Have you tried different boards? Can you try with other partitions (SPIFFS size)?

euphi, 2016-10-21:
I have only 512k Flash size, so I can't try a different size (`build_flags = -Wl,-Tesp8266.flash.512k64.ld`), It happens for different projects .

_**Note:** I use the ESP modules from in-circuit.de. They are of great quality and use a breadboad-friendly pin layout / grid.  Unfortunately they have only 512k flash size, resulting in various limitations, e.g.  I can't use OTA._

marvinroger, 2016-10-22:
I tried with an ESP01, two NodeMCUs V2, I can't reproduce. Can you try with a "bare minimum" sketch?

euphi, 2016-10-25:
Now its even more strange....

It seems to be broken during setup also outside of `homie.setup`?!

```C++
#include <Homie.h>

void setup() {
  Serial.begin(76500);
  Serial.println("Start");
  Serial.flush();
  Homie_setFirmware("bare-minimum", "1.0.0");
  Homie.disableResetTrigger();
  Serial.begin(76500);
  Serial.println("Setup");
  Serial.flush();
  delay(500);
  Serial.begin(76500);
  delay(500);
  Homie.setup();
  Serial.begin(76500);
  Serial.println("Finish");
  Serial.flush();
}

void loop() {
  Homie.loop();
}

```

results in output:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3ffe8480
~ld
lE���5
Setup
�&T�+k�5
✔ Wi-Fi connected
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_CONNECTED event...
Calling setup function...
Sending Wi-Fi signal quality (26%)...
Sending uptime (6s)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 1)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 2)...

```

You see, the initial "Start" string is missing - and the serial output recovers itself after some time without any call to `Serial.begin()`.

However, without the call to `Serial.begin()` directly before `Homie.setup()` the output does not recover?!

And, without calling `Homie.setup()` the serial output is fine, too?!

Maybe, something interferes with some ESP internal function, e.g. the Wifi driver. Maybe some interrupts distort the Serial clock?!

I think to analyse further a logic analyzer would be helpful (I don't have one yet).

DavidStacer, 2016-10-26:
Try using 74880 as the baud rate in your program.  I think that is the
default baud rate for the boot up messages.

Make sure your serial terminal matches the speed you set in the
Serial.begin();.

http://www.esp8266.com/viewtopic.php?f=33&t=2308

euphi, 2016-11-01:
Even more strange now:

This works fine now _(except the LED on PIN16, see below)_:

```C++
void setup() {
	delay(200);
	Serial.begin(74880);
	Serial.println("Starting Heizungscontroller");
	Homie.setLedPin(16, false);
	Homie.disableResetTrigger();
	LN.setLoglevel(LoggerNode::INFO);
	Homie_setFirmware("HzngCtrl", "1.0");
	Homie.setup();
}
```

If I remove `Homie.setLedPin(16, false);` the serial does not work. Because there is no `Serial.begin()` after Homie.setup()`, it also does **not** "start working again".

Oh, the LED itself does not work. It is always on, regardless if I set on to `true` or `false`.?!
( PIN 16 is connected on to the cathode of a standard red LED with the anode at a 90Ohm Resistor to 3,3V supply.)

The rest of Homie and my application works fine.

For completeness, serial output without `setLedPin()`:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3ffe84f8
~ld
Starting Heizungscontroller
�

< ... no more output ...>
```

With `setLedPin()`:

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3ffe84f8
~ld
Starting Heizungscontroller
Triggering NORMAL_MODE event...
** Booting into normal mode **
{} Stored configuration
  • Hardware device ID: f129afef
  • Device ID: hzgctrl
  • Boot mode: normal
  • Name: Heizungscontroller
  • Wi-Fi: 
    ◦ SSID: KDG-49D6B
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 192.168.0.220
    ◦ Port: 1883
    ◦ Base topic: devices/
    ◦ Auth? no
  • OTA: 
    ◦ Enabled? no
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected
Triggering WIFI_CONNECTED event...

< ...output continues ...>
```

marvinroger, 2016-11-01:
Got it. The default LED pin is `BUILTIN_LED` (https://github.com/esp8266/Arduino/blob/master/variants/generic/pins_arduino.h). On some board, the LED on pin 1 is also the serial line. So your serial was not working correctly because it conflicted with the LED blinking (when connecting to the Wi-Fi). Calling `Homie.setLedPin(16, false);` changes the LED pin, so Serial works back again. If you say the LED does not work, it probably means your LED pin is not actually on GPIO16. Are you sure the LED is on pin 16?

marvinroger, 2016-11-14:
@euphi was it the issue?

euphi, 2016-11-14:
I think so.
if I disable the LED or set it to another PIN it works fine.
It's still not clear why the LED did not work on PIN 16, but this is another issue.
Unfortunately I bricked my two last ESP modules and the others are in opearational use, so at the moment I don't have any to try.

I close this here and if I find an issue with the LED I will open a new issue here.

closed 2016-11-14 by euphi