homieiot/homie-esp8266 #158: Itead Sonoff doesn't connect to Wifi

pdominique, 2016-09-26:
I'm using Homie v2 and the example for the Itead Sonoff device. I know this example has not been tested but some people seem to have their Sonoff run fine with Homie. I'm using Platformio and I tried different board definitions (ESP8285 and ESP-01 with 1MB) but I'm not sure which one I should pick for this device.

I can flash the device but the led keeps blinking slowly so apparently it can't connect to my wifi network. I configured Homie with the web app and the HTTP API. I don't have any problem with Homie 1.5 on a Wemos D1 Mini on the same network.

pdominique, 2016-09-27:
Ok, I found a solution. Turns out you have to select the ESP-01 with 512k profile (even if the device really has 1M) to avoid this issue. Apparently you can also erase the flash memory to use the 1M (still need to confirm this): http://tech.scargill.net/itead-slampher-and-sonoff/#comment-13742

@marvinroger could we add this somewhere in the documentation maybe? And thanks for this great project by the way!

marvinroger, 2016-09-27:
This is an [esp8266/Arduino](https://github.com/esp8266/Arduino) problem more than an Homie one. :)
I've added a note: https://homie-esp8266.readme.io/v2.0.0/docs/limitations-and-known-issues

closed 2016-09-27 by marvinroger