homieiot/homie-esp8266 #159: Blocking mqtt

cingolanifede, 2016-09-26:
Hi. Is there any way to avoid blocking the code when mqtt connection get lost and cant reconnect?
Maybe using the handler in the loop function?

jpcacace, 2016-10-01:
hi...i 've been having the same problem....is there any way that the code doesn't block the app in the esp when the connection to the broker is lost??

euphi, 2016-10-01:
I never tried this with the current release 1.5, but the current development version 2 (that uses a new mqtt library) does not block when connection to mqtt broker is lost.

marvinroger, 2016-10-06:
Yes, the v1.x branch uses a sync MQTT implementations, the upcoming v2.x will fix this issue. I'll bump this topic when released.

closed 2016-10-06 by marvinroger

cingolanifede, 2016-10-08:
Thanks. Working now