homieiot/homie-esp8266 #160: DHT22 read failure

elle2, 2016-09-26:
Hi all, 
I'm trying to explore the "world of Homie" but I've a strange issue using DHT22 sensor.
If I read temperature value during loop function, with interval set to 300 seconds, it's ok.
Instead if I'm trying to read temperature after a received MQTT message, using a settable property and associated handler, it fails always.
Here is the sketch I'm using:
```
#include <Homie.h>
#include <DHT.h>

#define DHTPIN 4     // GPIO4  pin 2  of ESP8266`
#define DHTTYPE DHT22   // DHT 22  (AM2302)`
const int TEMPERATURE_INTERVAL = 300;
unsigned long lastTemperatureSent = 0;

HomieNode temperatureNode("temperature", "temperature");

DHT dht(DHTPIN, DHTTYPE);

void setupHandler() {
  Homie.setNodeProperty(temperatureNode, "unit").send("c");
}

float readTemp(){
    float t = dht.readTemperature();
    if (isnan(t)) {
          Serial.println("Failed to read temperature from DHT sensor!");
          return t;
    }
    Serial.print("Read Temperature: ");
    Serial.print(t);
    Serial.println(" °C");
    return t;
}
 
void getSendTemp() {
  if (millis() - lastTemperatureSent >= TEMPERATURE_INTERVAL * 1000UL || lastTemperatureSent == 0) {
    float t = readTemp();
    if (isnan(t)) {
          Serial.println("Failed getSendTemp!");
          return;
    }
    Homie.setNodeProperty(temperatureNode, "degrees").send(String(t));
    lastTemperatureSent = millis();
  }
}

bool tempReadHandler(HomieRange range, String value) {
  if (value == "true") {
      float t = readTemp();
      if (isnan(t)) {
          Serial.println("Failed ReadTemp!");
          return true;
      }
      Homie.setNodeProperty(temperatureNode, "read").send("true");
      Serial.print("Send Temperature: ");
      Serial.print(t);
      Serial.println(" °C");
      Homie.setNodeProperty(temperatureNode, "degrees").send(String(t));
      return true;
  } else {
      return false;
  }
}

void loopHandler() {
  getSendTemp();
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  dht.begin();
    
  Homie_setFirmware("my-dht22", "1.0.0");
  
  temperatureNode.advertise("read").settable(tempReadHandler);
  temperatureNode.advertise("unit");
  temperatureNode.advertise("degrees");
 
  Homie.setSetupFunction(setupHandler).setLoopFunction(loopHandler);

  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

For example if i send a message with 
`mosquitto_pub -t 'devices/012092e1/temperature/read/set' -m true`

the output is
```
Calling global input handler...
Calling node input handler...
Calling property input handler...
Failed to read temperature from DHT sensor!
```
during the loopHandler the output is
`Read Temperature: 25.60 °C`

Thank you

stone, 2016-09-28:
Just passing by, but I have had similar problems (not with Homie).
 
The datasheet on the DHT22 sensor states that you should not poll the 
sensor more than once every two seconds. The getSendTemp() has a 
check for that (every 3 sek). But if you do your mosquitto_pub -t  ..../set -m true
just after a getSendTemp() it often fails (tempReadHandler() is calling readTemp() 
without delay check).

elle2, 2016-09-28:
Thank you for reply, but the issue isn't correlated to reading frequency. 
Also if I remove the call to getSendTemp() in loop function, I've always the problem when I receive the MQTT message for topic "devices/012092e1/temperature/read/set"
```
Calling global input handler...
Calling node input handler...
Calling property input handler...
Failed to read temperature from DHT sensor!
Failed ReadTemp!
```
I think the problem is correlated to the behavior of the input handler function upon the reception of  MQTT message. Probably it's disabling something at low level, I suppose but I'm not able to analyze it.

snikch, 2016-10-04:
Hi @elle2 I'm seeing the same issue with a dht11. It works fine when I don't define a `HomieNode`, but the moment I do, it hangs while trying to read and then crashes after a few seconds. After reading your issue, I watched the logs for my mqtt broker and can see that the crash occurs when a message is published to the device.

```
Sending PUBLISH to homie-ota-6 (d0, q0, r0, m0, 'devices/test/$online', ... (5 bytes))
```

I don't need to do anything except simply instantiate a `HomieNode` to reproduce it. Just this one line:

```cpp
HomieNode temperatureNode("temperature", "temperature");
```

snikch, 2016-10-04:
@marvinroger Any thoughts on this one? Seems too much of a coincidence that we've both got a very similar problem.

snikch, 2016-10-04:
I just realised my mqtt log above is incorrect. That's publishing to the ota server. The lines before it are actually a timeout from my device.

```
1475572918: Client Fairholme-test has exceeded timeout, disconnecting.
1475572918: Socket error on client Fairholme-test, disconnecting.
```

I wonder if the sensor read is taking too long and causing the mqtt connection to time out.

marvinroger, 2016-10-04:
Which homie-esp8266 version are you guys using?

elle2, 2016-10-04:
Hi @marvinroger, I used homie-esp8266 in git repository. 
After you request I done a test using the stable release 1.5 and I can correctly read DHT22 sensor when I receive a MQTT message. 
Just a question: I rewrite my sketch to be compliant with release 1.5 documentation, for example I use  `Homie.setNodeProperty(temperatureNode, "degrees", String(t), true)` 
instead of
 `Homie.setNodeProperty(temperatureNode, "degrees").send(String(t))` used in my previous test.
Is it correct?
thank you

snikch, 2016-10-04:
I'm using homie via git as well (managed by platformio). Rev `bee6726`.

I'm happy to provide any debug info I can, let me know what you need.

marvinroger, 2016-10-07:
@me-no-dev can this be related to the readings done in an async callback?

nuccleon, 2017-04-20:
I have similar problems with v2.0.0 from the git-repository when reading DHT22 (adafruit/DHT-sensor-library). Reading DHT22 (within the loopHandler) **delays for more than 270ms**.
Without HOMIE it works well for several hours without reading errors.

So I've enabled DEBUG messages in the DHT library and I've seen always the same error: _"Timeout waiting for start signal low pulse."_
The library lock the interrupts during the reading process. I am not sure if the interrupt lock works as expected on the esp8266. They use `noInterrupts()` and `interrupts()`

benzino77, 2017-04-20:
Are you using Adafruit DHT library?

nuccleon, 2017-04-20:
Yes! (adafruit/DHT-sensor-library)

benzino77, 2017-04-20:
There is a piece of code:

```
boolean DHT::read(bool force) {
  // Check if sensor was read less than two seconds ago and return early
  // to use last reading.
  uint32_t currenttime = millis();
  if (!force && ((currenttime - _lastreadtime) < 2000)) {
    return _lastresult; // return last correct measurement
  }
  _lastreadtime = currenttime;

  // Reset 40 bits of received data to zero.
  data[0] = data[1] = data[2] = data[3] = data[4] = 0;

  // Send start signal.  See DHT datasheet for full signal diagram:
  //   http://www.adafruit.com/datasheets/Digital%20humidity%20and%20temperature%20sensor%20AM2302.pdf

  // Go into high impedence state to let pull-up raise data line level and
  // start the reading process.
  digitalWrite(_pin, HIGH);
  delay(250); // <-- LOOK HERE

  // First set data line low for 20 milliseconds.
  pinMode(_pin, OUTPUT);
  digitalWrite(_pin, LOW);
  delay(20); // <-- LOOK HERE
```
which explicit calls `delay` function. Have a look at the [exp8266/Arduino](https://github.com/esp8266/Arduino)  project documentation note about delays: [link](http://esp8266.github.io/Arduino/versions/2.3.0/doc/reference.html#timing-and-delays)

nuccleon, 2017-04-20:
Ok, `delay()` gives other tasks a chance to run. Sorry, but I don't see the problem. 
It also works fine without HOMIE.
The timing ciritcal section is protected with an interrupt lock. And within the lock, the use delayMicroseconds, that does not schedule other task.

```
{
    // Turn off interrupts temporarily because the next sections are timing critical
    // and we don't want any interruptions.
    InterruptLock lock;

    // End the start signal by setting data line high for 40 microseconds.
    digitalWrite(_pin, HIGH);
    delayMicroseconds(40);

    // Now start reading the data line to get the value from the DHT sensor.
    pinMode(_pin, INPUT_PULLUP);
    delayMicroseconds(10);  // Delay a bit to let sensor pull data line low.

    // First expect a low signal for ~80 microseconds followed by a high signal
    // for ~80 microseconds again.
    if (expectPulse(LOW) == 0) {
      DEBUG_PRINTLN(F("Timeout waiting for start signal low pulse."));  <-- THROWN ERROR
      _lastresult = false;
      return _lastresult;
```

benzino77, 2017-04-20:
Oh ... I just want to show you where those 270ms comes from and what happens during `delay` :/
AFAIK you cannot disable all interrupts on esp8266 so it is possible that task which are timing critical will return bad result.
Have you tried other libraries for DHT22?

nuccleon, 2017-04-20:
No, I didn't try another DHT22 library. All the libs that I know read the sensor in the same way (with some deviation in the delays).
`expectPulse` returns 0 (my error pattern) either 

1. the data pin has an unexpected level or
2. the level is as expected for more than `_maxcycles`

```
  _maxcycles = microsecondsToClockCycles(1000); 

    while (digitalRead(_pin) == level) {
      if (count++ >= _maxcycles) {
        return 0; // Exceeded timeout, fail.
      }
```

Secenario 1. could be the case if `delayMicroseconds(10);` within the locked region takes longer than 80us (unexpected interrupt????)

```
    // Now start reading the data line to get the value from the DHT sensor.
    pinMode(_pin, INPUT_PULLUP);
    delayMicroseconds(10);  // Delay a bit to let sensor pull data line low.

    // First expect a low signal for ~80 microseconds followed by a high signal
    // for ~80 microseconds again.
    if (expectPulse(LOW) == 0) {
```

The main question is:
Why does the same code work without using Homie?
Is there a higher (unlockable) interrupt load after instantiation of a HomieNode?

benzino77, 2017-04-20:
Have you tried to use other PIN of ESP8266? Also try to use:
```
Homie.disableLedFeedback();
Homie.disableResetTrigger();
```
before Homie.setup();

amayii0, 2017-04-21:
Hi all, I've been following this thread for some time now.
Long enough for me to finalize this post at instructables: https://www.instructables.com/id/Building-Homie-Devices-for-IoT-or-Home-Automation/
I've included schema and code for DHT22, DS18B20 and a photosensitive module.
Code for DHT22 is here : https://github.com/amayii0/Homie-DHT22

All work fine for me but however I occasionally get failures reading DHT22 temp. However I haven't had a look if that was caused by reading temp, humidity or both, see https://github.com/amayii0/Homie-DHT22/blob/master/src/main.cpp#L31

Something wrong regarding pull-up on data line maybe?

nuccleon, 2017-04-21:
@benzino77 I'm going to test your suggestion tonight ...
@amayii0 There is a pull-up on the DHT shield for the arduino. And the Datafruit DHT library configure the data line as INPUT_PULLUP. This might be a problem if the total resistance of the parallel resistors is too low.

Anton-V-K, 2017-09-11:
If this issue isn't resolved yet, take a look at https://github.com/adafruit/DHT-sensor-library/issues/48

stritti, 2019-03-23:
https://github.com/adafruit/DHT-sensor-library/issues/48 is solved since Feb 14. 
@elle2 could you retest now?

closed 2019-03-30 by stritti