homieiot/homie-esp8266 #162: total IOT home solution

mkeyno, 2016-09-30:
sorry  @marvinroger to draw ticket for my question , actually I read your homie documents but honestly I don't know whether homei can totally server my IOT home project or not , but consider there is couple of sensor(temp, humid...) and actuator (relay, buzser,..) in home that should 
report back its status  to the server( in that case raspberry pi) or
get command from server or 
sensor command related actuator while report its activity to server 

do I get all this configuration and data  for what added in the device networks? 
if you know links to such practice that  use  raspberry pi , really appreciated to share it

clough42, 2016-10-03:
I would suggest you check out Node-Red:  http://nodered.org/

For my home automation, I have a number of devices using homie-esp8266.  They all connect to an MQTT broker.  The devices don't know about each other.  I use Node-Red to wire everything together.

marvinroger, 2016-10-07:
As @clough42 said, your Raspberry Pi will run an MQTT broker, and the devices will connect to the broker to report their status, depending of what your sketch contains. Feel free to look at the examples projects on the repo.

mkeyno, 2016-10-08:
@marvinroger  has anyone yet use your lib with Raspberry Pi? I would like take your lib as my IOT home project and any links or information to real practice really appreciate

marvinroger, 2016-10-08:
`homie-esp8266` runs on ESP8266 exclusively. But yes, a lot of people here use a Raspberry Pi as a MQTT broker.

Nzbuu, 2016-10-08:
There's also a `homie-python` project that is equivalent to `homie-esp8266` for Raspberry Pi, etc. I don't know how mature it is though.

mkeyno, 2016-10-08:
thanks @Nzbuu , do you know any link for that ?

Nzbuu, 2016-10-08:
https://github.com/jalmeroth/homie-python

ordonot, 2016-10-12:
@mkeyno 
If you want a "total IOT home solution" you may need more than MQTT. My home is driven by openHab (+Mosquitto +Homie devices +some-other-stuff). Basic setup is described in this [colorful blog post by RichP](http://www.wemustbegeeks.com/esp8266-send-dht-temperature-humidity-mqtt-oled-display/).

rplakas, 2016-10-12:
@mkeyno @ordonot Someone mentioned my name...am I in trouble again ;-)

I currently use OpenHab to centralize my Home Automation but Home Assistant is another one to look at. Both are open source . I tried Home Assistant briefly but it didn't support my alarm at the time (DSC Alarm) but now it does. OpenHab has an advantage for me with their app for out of home notifications and control. I believe Home Assistant requires you to open ports to the outside. I'm working on a ESP8266 Homie/Cayenne Garage Door opener at the moment.

euphi, 2016-10-24:
Regarding Openhab2 (and thus Eclipse SmartHome) a discussion has been started at https://github.com/marvinroger/homie/issues/10 .

Bee-Certain, 2016-10-26:
So many options, so little time.

I'm still getting started with Homie.  I use:
panStamp-->LagartoSWAP-->Mosquito-->NodeRed-->InfluxDB-->Grafana.
and
NodeRed-UI-->NodeRed-->NodeRed-UI

I will be adding:
Homie-->Mosquito-->Homie
to the above mix.

Sorry if the above is confusing, it's just my way of saying that I use InfluxDB and Grafana for heavy graphing and the NodeRed-UI for any custom user interface items such as switches, temperature displays, etc.

Mosquito should make Homie integration a snap.  As I said, I'm just getting started with Homie.

Bob

closed 2019-03-23 by stritti