homieiot/homie-esp8266 #163: Is there a way to send LWT?

diazleonardo, 2016-10-02:
If there is not, may I suggest adding it? I quickly went through the MQTT [standard](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718037) and found no mention to LWT.  Is a description of what, and when to send the LWT somewhere?  

If this helps on the implementation, the signature for different clients is as follows:

paho-mqtt:
    `will_set(topic, payload=None, qos=0, retain=False)`

NodeMCU client:
   `mqtt:lwt(topic, message[, qos[, retain]])`

IBM WebSphere MQ Telemetry:
   `MqttConnectionOptions.setWill(MqttTopic lastWillTopic, byte [] lastWillPayload, int lastWillQos, boolean lastWillRetained)`

benzino77, 2016-10-02:
homie will leave LWT in `devices/<id>/$online` topic. When it not cleanly disconnect from broker (i.e. network failure) broker will set this topic to `false`. 

You can read more about lwt here: 
[http://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament](url)

diazleonardo, 2016-10-03:
Fair enough,  but is there a way where I can set the topic and message for the LWT?  There are many reasons for going offline,  in my especial case, deep sleep.

benzino77, 2016-10-04:
In version 2.0 there is new functionality. Have a look at [here](https://homie-esp8266.readme.io/docs/deep-sleep) 

Yo can also try to define your own LWT by getting acces to underlying homie mqtt client: [mqtt-client](https://homie-esp8266.readme.io/docs/miscellaneous#section-get-access-to-the-mqtt-client), but I'm not sure if it won't broke homie itself. 
Here is the link to async-mqtt-client used by homie [async-mqtt-client](https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md#asyncmqttclient-setwillconst-char-topic-uint8_t-qos-bool-retain-const-char-payload--nullptr-size_t-length--0)

marvinroger, 2016-10-07:
LWT is defined in the spec [here](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Will_Flag).

Only one LWT can be defined, and, following the Homie convention, `homie-esp8266` defines one as @benzino77 explained. So yes, you can use the `prepareForSleep()` function which will take care of cleaning the MQTT connection so you can deep sleep.

closed 2016-10-25 by marvinroger