homieiot/homie-esp8266 #165: Button to trigger relay locally (on a Sonoff)

n8twj, 2016-10-06:
Hi folks -  I have several Sonoff devices now running Homie (git head version).    One thing I have not figured out how to do is to interact with the onboard momentary switch... to trigger the relay / force a reboot into config mode / etc  

https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Button-usage has several options using multiple short / long presses - which seems potentially useful. 

Here is my sketch that works with an attached DHT22 (AM2301) to pin 1 of the header (GPIO14) 

`#include <Homie.h>
#include "DHT.h"

#define DHTPIN 14 
#define DHTTYPE DHT21

const int PIN_RELAY = 12;
const int PIN_LED = 13;
const int PIN_BUTTON = 0;
const int INTERVAL = 60;

unsigned long lastSent = 0;

DHT dht(DHTPIN, DHTTYPE);

HomieNode temperatureNode("temperature", "temperature");
HomieNode humidityNode("humidity", "humidity");
HomieNode switchNode("switch", "switch");

bool switchHandler(HomieRange range, String value) {
  if (value == "true") {
    digitalWrite(PIN_RELAY, HIGH);
    Homie.setNodeProperty(switchNode, "on").send("true");
    Serial.println("Switch is on");
  } else if (value == "false") {
    digitalWrite(PIN_RELAY, LOW);
    Homie.setNodeProperty(switchNode, "on").send("false");
    Serial.println("Switch is off");
  } else {
    return false;
  }

  return true;
}

void setupHandler() {
  Homie.setNodeProperty(temperatureNode, "unit").setRetained(true).send("F");
  Homie.setNodeProperty(humidityNode, "unit").setRetained(true).send("percent");
}

void loopHandler() {
   if (millis() - lastSent >= INTERVAL * 1000UL || lastSent == 0) {
     float humidity = dht.readHumidity();
     float temperature = dht.readTemperature(true);
     Homie.setNodeProperty(temperatureNode, "F").send(String(temperature));
     Homie.setNodeProperty(humidityNode, "F").send(String(humidity));
     lastSent = millis();
  }
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW);

  Homie_setFirmware("itead-sonoff", "1.0.0");
  Homie.setLedPin(PIN_LED, LOW).setResetTrigger(PIN_BUTTON, LOW, 5000);
  
  Homie.setSetupFunction(setupHandler);
  Homie.setLoopFunction(loopHandler);

  switchNode.advertise("on").settable(switchHandler);
  temperatureNode.advertise("unit");
  temperatureNode.advertise("temperature");
  humidityNode.advertise("unit");
  humidityNode.advertise("percent");

  Homie.setup();
}

void loop() {
  Homie.loop();
}`

marvinroger, 2016-10-07:
Just like any other sketch, you can debounce the button in the `loop()`, if you detect a short press you toggle the relay and if you detect a long press you can:

```cpp
Homie.eraseConfiguration();
ESP.restart();
```

Really, think of it as a normal Arduino sketch. 😉

n8twj, 2016-10-07:
Ok - But don't I have to tell Homie that the relay status has changed?

marvinroger, 2016-10-07:
If you change manually the status of your relay, just call `Homie.setNodeProperty(relayNode, "on").send("true --or-- false");` to tell the broker the actual relay now has a new state.

n8twj, 2016-10-07:
Perfect - I will work on that now :)      

I have several sensors operational with Homie https://github.com/n8twj?tab=repositories  several more to come  :)

marvinroger, 2016-10-07:
Awesome, thanks for sharing!

marvinroger, 2016-10-25:
There's now an example: https://github.com/marvinroger/homie-esp8266/blob/develop/examples/IteadSonoffButton/IteadSonoffButton.ino

closed 2016-10-25 by marvinroger