homieiot/homie-esp8266 #166: My activity LED on GPIO0 doesnt want to work

claybar, 2016-10-07:
I am using the head of 2.0 branch with a 'LoLin' NodeMCU clone.  I have attached an LED+resistor between to GPIO0 and 3V3.  Everything else is working (MQTT messages firing nicely), but the LED only emits a couple of short pulses after boot, then nothing.

```
Homie_setFirmware(FW_NAME, FW_VERSION);
Homie.disableResetTrigger();
Homie.setLedPin(D3, LOW);
Homie.setSetupFunction(homieSetupHandler);
Homie.setLoopFunction(homieLoopHandler);
Homie.onEvent(onHomieEvent);
Homie.setup();
```

The D3 define maps to GPIO0, and I have made it active low so the boot pull-up resistor is not defeated by the LED.

Should this work?

benzino77, 2016-10-07:
Homie uses LED to show the state of itself: [docs](https://homie-esp8266.readme.io/docs/getting-started#section-understanding-what-happens-in-normal-mode)
So if your NodeMCU connects to Wifi and MQTT fast, then you are able to see only couple of flashes.

marvinroger, 2016-10-07:
@claybar does it answer your question?

claybar, 2016-10-07:
Thanks, that answers it.  I was expecting the LED to show activity as well
as status.  I was under the (false) impression the LED would also flash
whenever MQTT messages are sent.  But seems to be working as advertised.

Is there a way the LED could also be used for activity?  I guess I could
drive the LED directly from my code, but that seems a bit dangerous and
could conflict with Homie's control of the LED.

claybar, 2016-10-07:
Also, I suggest a link is placed in the 'Build-In LED' section to the
description of what the LED displays.

marvinroger, 2016-10-08:
@claybar you can drive the LED directly from your sketch. As long as you drive the LED from the setup handler, the loop handler and/or inside an `isConfigured()`/`isConnected()` if block, there won't be any conflict as these part of the code won't be called when Homie will take control of the LED.

[I updated the docs](https://homie-esp8266.readme.io/docs/built-in-led). Is it okay?

claybar, 2016-10-08:
Thanks for the info.  Docs link good.

closed 2016-10-08 by claybar