homieiot/homie-esp8266 #167: Cant get Homie to work sucessfully

neilstreet6036, 2016-10-08:
Hi,

Im trying to use Homie but without success. I have tried this very basic sketch 

#include <Homie.h>

void setup() {
  // put your setup code here, to run once:
  Homie.setup();
}

void loop() {
  // put your main code here, to run repeatedly:
  Homie.loop();
}

When i upload using Arduino IDE 1.6.11 directly or via a .bin file through ESPFlasher i get the following error. I have tried several ESP12e's and no luck, I can flash other firmware etc onto the devices without any issues. Any ideas please?? I have tried different versions of the IDE from 1,.6,.5 upward.

Neil.

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

marvinroger, 2016-10-08:
Can you add `Homie.disableResetTrigger();` and `Homie.disableLedFeedback();` before `Homie.setup()`?

neilstreet6036, 2016-10-08:
Hi,

Added the lines as suggested and still get this error

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

I have downloaded your latest build and updated all other libraries

benzino77, 2016-10-08:
> #include

This is just typo?

benzino77, 2016-10-08:
Do you have proper board chosen?
Tools -> Board -> NodeMCU 1.0 (ESP-12E module)

neilstreet6036, 2016-10-08:
Yes, I have the correct board selected (and yes its a typo the line reads #include <Homie.h>)

marvinroger, 2016-10-10:
#135 was similar to this issue
Can you try changing the SPIFFS size, and eventually try with another board setting?

marvinroger, 2016-10-10:
And power cycle your ESP8266 after the flash, it might help.

n8twj, 2016-10-10:
I too had this problem... First off, verify you are running a solid/known good FTDI adapter.  I believe this was easily 50% of my troubles...I bought a lot of 5 cheap ones from [insert random shenzhen clone company].... it seems at least two of them (so far) have became flakey after some [ab]use.
I would get various errors during upload and/or immediate exceptions endless rebooting on the ESP) [prolly due to corrupted uploads]

Then, if you are using a plain ESP-12(e) module in your own design/breakoutboard/pcb, verify your wiring against this:  https://github.com/esp8266/Arduino/blob/master/doc/ESP_to_serial.png - and then start with a 512k flash size (64k SPIFFS) - You can increase later, if second step works.  

Next, I always start with loading a simple  'hello world'... setup() { Serial.begin(115200); } loop {  Serial.Print("Hello World!"); } sketch...   [This step almost seems required in several cases...I have noticed a trend (in at least via the "cheap" sources I have utilized) that NodeMCU with LUA firmware does not like to immediately cooperate with the Arduino based firmware...needs more testing...]

[Sidenote:  If I am using a device that uses i2c based peripherals, I add a third step of loading an i2c scanner sketch: http://pastebin.ca/3727270] 

Then I load Homie  - Typically using a 4M Flash (3m SPIFFS) on ESP-12[e] based modules/designs. 

And yes - a power cycle after the initial flashing of Homie seems to be required.  

Good luck -  Post more details as you continue to have trouble  :)

marvinroger, 2016-10-14:
@neilstreet6036 any updates?

neilstreet6036, 2016-10-15:
Hi, Sorry been away on business most of the week. I have replaced the FTDI with one i ordered earlier in the week from ebay, genuine uk part not an import - same result. I have tried the various SPIFFS settings, same result. I am now in the progress of assembling an new test board with a new esp12f. I should have something by this evening in terms of success/failure.

neilstreet6036, 2016-10-16:
Some success! 

Ordered a new esp12f , built a new proto board and whilst testing the connections and voltages i noticed that the breadboard power supply adapter was only kicking out 2.6V, arrrrghhh! connected a new adapter to the original board and now some success. The esp boots into config mode but when i try to use the web app to configure it I get the following message. 

UI bundle not loaded. See Configuration API usage: https://homie-esp8266.readme.io/docs/http-json-api

n8twj, 2016-10-16:
You are getting close...   Connect to the internet, go here http://marvinroger.github.io/homie-esp8266/  then switch to the Homie-XXXXX Wifi -  the app should then react and allow you to configure

neilstreet6036, 2016-10-18:
Took a long while to connect to the AP but i think its now configured. Any examples of some simple mqtt tests to try just to make sure?

marvinroger, 2016-10-18:
Just check the Serial monitor. If it's written you're connected to MQTT, then you're for sure connected as it means the `CONNACK` packet is received. Or try to send a message to `<base_topic>/<device_id>/test/test/set`, you will see in the Serial monitor that the `test` node does not exist... So you'll see it correctly receive messages.

neilstreet6036, 2016-10-18:
Looks like im back to square 1. I have flashed two devices with the firmware ok, connected to the Homie AP's, configured through the the web interface and now once they are rebooted both report this message through the serial port monitor

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

Where to go from here??

marvinroger, 2016-10-18:
After you flash, reboot your ESP8266 before configuring it. There's a known esp8266/Arduino issue that trying to reboot the ESP8266 from the code (like we do) triggers a `boot mode:(3,7)` right after a flash. But if you power-cycle the device once, it works fine.

neilstreet6036, 2016-10-19:
ill give it a try and feedback on the results. Thanks.

neilstreet6036, 2016-10-19:
here's what i did.

Flashed the esp.
Removed the power cable and the usb cable to the programming board to ensure there was no chance of any power to the esp. 
Reconnected the power and usb.
Loaded the homie config page.
Disconnected from my router (192.168.1.1)
Connected to the homie ap.
Configured the node and a successful configuration message was received.
Power cycled the esp.
Opened the serial monitor in the arduino ide
Nothing displayed,
Reset the esp
And... the same reset ,message on the serial monitor.

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

Im really keen on using the homie as it appears to be a great node but cant understand why the issues are still occurring.

marvinroger, 2016-10-25:
Ok, please pull the latest version and flash this:

```c++
#include <Homie.h>

void setup() {
  Homie_setFirmware("bare-minimum", "1.0.0");
  Homie.disableResetTrigger();
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

And report everything that's going on in your serial monitor.

twinkl01, 2018-01-16:
Hello, I have the same issue. In browser I get the following message:
UI bundle not loaded. See Configuration API usage: https://homie-esp8266.readme.io/docs/http-json-api

I tried everything n8twj wrote. And I pull the latest version (Branch: develop) with 

`#include <Homie.h>

void setup() {
  Homie_setFirmware("bare-minimum", "1.0.0");
  Homie.disableResetTrigger();
  Homie.setup();
}

void loop() {
  Homie.loop();
}`

The report on serial monitor with baud 74880 is

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

Thank you maik for your great work.

best regards Nick

timpur, 2018-01-16:
firstly you might need to add `Serial.begin(115200);` (http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.3/quickstart/getting-started/) as the start of `setup()`.

also see http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.3/advanced-usage/ui-bundle/ to get the ui bundle.

euphi, 2018-01-18:
BTW: I prefer to set the baudrate to 74880 so it is the same as the boot loader. 

`Serial.begin(74880):`

Although quite uncommon, nowadays most terminal programs can handle this baudrate.

Furthermore I always send a short message and flush it, to be sure, if  `setup()` has been reached.

```
void setup() {
  Homie_setFirmware("bare-minimum", "1.0.0");
  Serial.begin(74880);
  Serial.println("Start");
  Serial.flush();
  Homie.setLoggingPrinter(&Serial);
  Homie.setup();
}
```

If you use libraries or have global objects please be aware that constructors are called before `setup()`, so if setup() is not reached, the cause for a problem is most likely in one of the constructors.

luebbe, 2018-01-18:
Same here. I also prefer 74880. I don't know why 115200 got so popular. Is this the arduino default whereas the ESP default is 74880?
I also define it in platformio.ini, like so:

```
[common]
serial_speed = 74880

[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
...
build_flags = -D SERIAL_SPEED=${common.serial_speed}
...
monitor_baud = ${common.serial_speed}
monitor_rts = 0
monitor_dtr = 0

```
This ensures that I always have the same serial speed defined in my running application and in my serial monitor.
Some libraries unfortunately redefine serial speeed to 115200 for their specific logging.

closed 2019-03-30 by stritti