homieiot/homie-esp8266 #168: **Native english speakers**: ready for sleep or ready to sleep?

marvinroger, 2016-10-08:
I have a quick question for you guys native english speakers. Is `prepareToSleep()` or `prepareForSleep()` better? Or even `prepareForSleeping()`? Same for `READY_TO_SLEEP`, `READY_FOR_SLEEP` or `READY_FOR_SLEEPING`? Thank you!

jpmens, 2016-10-08:
I think `prepareToSleep()` is better.

I am tired and ready to go to sleep.
I am tired and ready for bed.

marvinroger, 2016-10-08:
That's what I thought, but I just saw this: [HOW TO PREPARE FOR SLEEP WITH MINDFULNESS](http://www.dummies.com/religion/spirituality/how-to-prepare-for-sleep-with-mindfulness/), so I was wondering.

jpmens, 2016-10-08:
The first thing I noted was the URL which contains, in order, the words "dummies", "religion", and "spirituality". Nothing against these of course....

Furthermore, that text says `or being on the computer before bed`. To be honest, I've spent years and years with computers, but have never actually been _on one_. #facepalm

marvinroger, 2016-10-08:
Good point, but the *dummies* books are serious and well-known in France, I thought the above link was the english equivalent. Thanks!

claybar, 2016-10-10:
A bit late off the mark, but my vote would also be for `prepareToSleep()`.  A natural follow-on would be `goToSleep()`, the alternative to this `goForSleep` sounds weird.

Nzbuu, 2016-10-10:
I think:

* prepareToSleep means prepare for the act of going to sleep, "sleep" here being a verb
* prepareForSleep means prepare for the state of being asleep,  "sleep" here being a noun

So they're both OK,  but for me the first is slightly more natural.

marvinroger, 2016-10-10:
Alright, thank you! We all agree for `prepareToSleep` and `READY_TO_SLEEP`. 👍

closed 2016-10-10 by marvinroger