homieiot/homie-esp8266 #169: Upgrade homie-esp8266 1.5.0 to 2.0 via OTA?

jalmeroth, 2016-10-11:
Hi @marvinroger,

will homie-esp8266 v2 be in a way compatible so we can upgrade from v1.5.0 via OTA?

Thanks & regards,
Jan

marvinroger, 2016-10-11:
Well, no, the only problem being the `config.json` file changed. Sorry!

jalmeroth, 2016-10-12:
You mean missing key/values or what is going to break? Wouldn't it be possible to 'upgrade' them or at least populate with default values?

marvinroger, 2016-10-17:
I talked a bit too fast. It *will work* (without guarantee) if:

* You don't use SSL for MQTT
* Yon don't use mDNS

But you would better try with a dev ESP8266 on which you flash homie-esp8266 v1.5 with a similar config. If you have a device in an hard to reach place, you don't want to crash it by sending an OTA payload that would not be backward compatible. In the worst scenario, the config won't be recognized and the device will reboot in `config` mode, though.

jalmeroth, 2016-10-17:
Thank you ver much, @marvinroger!

I successfully upgraded one of my devices from homie v1.5.0 to v2 via OTA. 👍

closed 2016-10-17 by jalmeroth