homieiot/homie-esp8266 #170: Need to include another library

violachie, 2016-10-12:
Hi,

There are two questions i would like to ask:
1. When i write the minimum code (only include Homie.h), i always have error message that i have to include another libraries (ESP8266WiFi.h, PubSubClient.h, etc). After i include it all, there are no error message (I use Arduino IDE). Do i miss something during my installation step?

2. When i configure the homie mqtt, when enter the mqtt credentials, and click next button. It is still in mqtt credentials page instead of to the next page, like it has no respond.

I am new to this iot world, sorry if the question is so trivial. Thanks.

benzino77, 2016-10-12:
Did you install homie in library folder in sketch folder location?
Skech folder location can be obtained from File->Preferences. There should be subfolder libraries and there should be homie located (and other needed libraries).

violachie, 2016-10-12:
I find the homie libraries in folder Arduino/Portable/sketchbook/libraries/homie-esp8266-1.5.0
Same place with ArduinoJson, PubSubClient, etc (all the dependencies for homie mqtt), but i still need to include it in the code.

benzino77, 2016-10-12:
But all dependecies are in `Arduino/Portable/sketchbook/libraries` and not in `Arduino/Portable/sketchbook/libraries/homie-esp8266-1.5.0` ?

violachie, 2016-10-12:
The other dependencies library is in Arduino/Portable/sketchbook/libraries. So homie-esp8266-1.5.0 and PubSubClient, etc are in the same folder, Which is automatically located there by my Arduino IDE when I add .zip or via "manage libraries" menu.

marvinroger, 2016-10-13:
What's your Arduino version?

violachie, 2016-10-13:
I use wemos d1 mini

marvinroger, 2016-10-13:
I mean the version of your Arduino IDE.

violachie, 2016-10-13:
Oh sorry, now i am far from my computer. But I downloaded it about two months ago from Arduino website. I will confirm the version of my IDE as soon as I get home.

violachie, 2016-10-14:
Hi @marvinroger 

My arduino IDE version: 1.6.5

This is the folder where homie mqtt libraries are located
[image]

And this is the my "minimal" code for homie mqtt to run

`#include <Homie.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>
#include <Ticker.h>
#include <ESP8266mDNS.h>
#include <Bounce2.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <ESP8266httpUpdate.h>
#include <ESP8266HTTPClient.h>

void setup() {
  Homie.setup();
}

void loop() {
  Homie.loop();
}`

marvinroger, 2016-10-14:
This is indeed not written in the getting started guide, but library to library dependencies was added in Arduino 1.6.6. Update your IDE to the latest version and you'll see it works. The 1.6.5 is a year old.

violachie, 2016-10-14:
Oh, i see. Now i wonder where did i download arduino IDE from. Thanks for the answer, i will download the latest version.

closed 2016-10-14 by violachie