homieiot/homie-esp8266 #171: OTA: Status/progress reporting over MQTT

mrpace2, 2016-10-13:
Since this is my first contribution to Homie, I would like to start by thanking you, Marvin, for the great job you are doing with Homie.

I think v2 OTA over MQTT is a big improvement compared to v1 OTA over HTTP. I am using the latest v2 homie from git and @jpmen's homie-ota, and it worked out of the box. However, I think the Homie OTA definition is not yet complete. I would like to suggest a few enhancements to be considered for the v2 release. This issue covers one such improvement. I will submit a few more issues containing additional proposals.

One thing that I believe is missing is that Homie publishes feedback during OTA. Status updates would enable the OTA server to display progress and to report OTA results to the user and to log OTA results. Currently, when you click "GO!" in homie-ota, you get a new page saying "updating..." or so, and that's it. homie-ota can't tell what the firmware is up to. There's no feedback for whether or not the firmware update actually triggered (firmware version check, firmware running in normal mode...) or if the update is complete or still ongoing and whether it succeeded or failed.

BootNormal::_onMqttMessage could periodically post to a new topic `$implementation/ota/status`. Message format could be JSON `{"state":"...","info":"..."}` or, if plain text messages are preferred, two topics could be used (`$implementation/ota/state` and `$implementation/ota/info`).

Values of `<state>` and `<info>` could be:

Event | `<state>` | `<info>` | Description
----- | --------- | -------- | -----------
publish to `$ota`, different firmware version | "flagged" | "1.0.1" | New firmware version number
publish to `$ota`, identical firmware version | "success" | "same version" | Firmware version not changed
publish to `$implementation/ota/payload`, update is ongoing | "updating" | "8192/342512" | Format: `<bytes done>/<bytes total>`
publish to `$implementation/ota/payload`, update is complete | "success" | "firmware updated" | Firmware update successful
on error | "error" | "some error message" | Error during firmware update

Alternatively, a single $implementation/ota/status topic could use an "HTTP-like" convention:

`<message>` | Description
----------- | -----------
200 Success | Firmware update successful
206 `<bytes done>`/`<bytes total>` | Update in progress
304 Not Modified | Requested $ota version same as running firmware
400 `<error message>` | On any error

I guess I'd prefer the latter. I can work on a pull request if this proposal makes sense.

marvinroger, 2016-10-13:
Thanks a lot, Frank. Glad you like the project!

OTA Status update would definitely be a great addition, allowing some great GUI to be made.

I indeed prefer the latter too. However, we should normalize the status codes. I mean, in your table, `200` and `304` both have a status code text, whereas `206` and `304` don't. I know this is a valid this is valid in HTTP, but I think we should either set a status code text for every status, or not at all. I would prefer the second approach, without status text, as pretty much every developer knows `2xx, 3xx`... I would also differentiate client and server errors, with a `500` error code. The `400` error code would mean something is wrong in the payload for example (too big...) and the `500` error code would mean there was an internal problem, like an error during the flash.

mrpace2, 2016-10-13:
I'm fine with sending HTTP-like status codes only. This should simplify processing on the server side, even  though I think server-side will have enough resources to do that. Also, great idea to distinguish between 4XX and 5XX.

The only "features" we would miss are progress information during update and human-readable error messages. Any thoughts on that? Maybe we should keep standard status text with all responses, to be consistent, and append any variable data (byte progress with 206, actual error message with 4XX, 5XX) after a line feed or so? Or use a second topic for such extra information?

marvinroger, 2016-10-14:
I mean, the `status` topic could look like this:

* `200`
* `206,100/300000`
* `304`

Only a status code followed by a separator, and a special payload related to the status code. What do you think?

mrpace2, 2016-10-14:
I'd add these two:

- 400,firmware image too large
- 500,cannot write flash

(actual error messages after comma may vary). Or did you leave out 4XX and 5XX on purpose? 

Shall I submit a pull request? Based on the latest develop branch?

marvinroger, 2016-10-14:
It was only to show how it would look like. Of course, the list was not exhaustive. 😉

marvinroger, 2016-10-14:
If you have time to, then I'd appreciate a PR!

marvinroger, 2016-10-17:
Fixed by #174 thanks to @mrpace2

closed 2016-10-17 by marvinroger