homieiot/homie-esp8266 #173: OTA: Allow for updating SPIFFS

mrpace2, 2016-10-13:
For Homie projects using SPIFFS, it would be beneficial to also allow for updating the filesystem remotely.

esp8266/Arduino Updater provides support for SPIFFS updates. We just need a Homie convention for OTA/SPIFFS. Maybe use topics `$implementation/ota/firmware` and `$implementation/ota/filesystem` instead of `$implementation/ota/payload`?

Depending on the topic, `BootNormal::_onMqttMessage()` would specify `U_FLASH` or `U_SPIFFS` to the `command` argument when calling `Update.begin()`.

marvinroger, 2016-10-13:
The problem is the `config.json` and some state file (e.g. `BYPASS_STANDALONE` are actually stored in the SPIFFS. So if we overwrite it, the device would basically be "factory resetted".

mrpace2, 2016-10-13:
Oops. I didn't think about `config.json` and didn't even know about the state file. I guess it's too risky (e.g. power fail during update) to temporarily keep those in RAM while updating SPIFFS and then copying them back into the new SPIFFS. 

Let me review the SPIFFS code and the Updater code. Maybe we can "mount" the Updater memory range as a temporary SPIFFS partition and copy the two files into this partition after (or right before) calling `Update.end()`. That way, even if the SPIFFS upgrade fails, there would always be a good SPIFFS partition with a valid configuration.

mrpace2, 2016-10-18:
Working on #172 I learned that the esp8266/Arduino `Updater` class updates SPIFFS "in-place". That is, it writes the update directly into SPIFFS's flash sectors. So there is no second copy of SPIFFS available that we could have used for mounting a "shadow SPIFFS" to backup Homie's SPIFFS files before committing the SPIFFS update to flash. 

I think it would be too much hassle to mimic for SPIFFS inside Homie what the `Updater` class is doing for firmware. If at all, this feature should be implemented inside `Updater`. Maybe one day I will submit a PR to the esp8266/Arduino `Updater` class.

Looks like this wasn't the greatest feature proposal ;-). Would you mind if I closed this issue?

marvinroger, 2016-10-18:
Thanks for your time! Go ahead and close the issue :) we can reopen it if it's ever available in `Updater`.

closed 2016-10-18 by mrpace2